realsenseai/librealsense #2563: Long rosbag recording causes computer to freeze/crash

Vivikyung, 2018-10-18:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 5.10.03 | 
| Operating System & Version |    Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |  16.04                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.0                   | 
| Language                            |  C++                          |

### Issue Description

I was able to get the recording and playback features working as noted [here](https://github.com/IntelRealSense/librealsense/tree/development/src/media). Essentially, I am recording the depth and RGB streams to a rosbag for later post-processing.

However, I noticed that typically, after recording for about 7.5 minutes or so, my computer (screenshot below) will freeze and crash. These 7.5 minute rosbag videos seem to be ~15.4 GB. 
I need to be able to record ~30-60 minutes of video in one go, to be processed later. At least, I will need the RGB color frames and the depth information corresponding to them -- the video itself doesn't actually matter. However, I also cannot compromise on the FPS: I tried writing both streams' frames to files continuously, which worked but resulted in very low FPS.

Are there any solutions for this? Whether that be a way for the rosbags to not freeze my computer when recording for more than 7.5 minutes, or a different way altogether? I'm fine if I need to write my own script, but I don't even know which functions to start with.
Thank you!

[image]

dorodnic, 2018-10-18:
Hi @Vivikyung 

As far as I know, there should be no limitation on recording size. We will try to reproduce (it is not unlikely we unintentionally introduced some new bug causing this) and update.
However, the team worked hard to make sure this type of use-case would perform and even [contributed](https://github.com/ros/ros_comm/pull/1223) some of our findings about large ROS-bags back to the ROS community. 

If you prefer to avoid the built-in recorder, you can save data directly to disk. In that case:
a. Make sure you write from a dedicated thread and maintain large enough frame-queue (and call `frame.keep()` when enqueening)
b. Make sure the SDK is compiled with optimizations (Release configuration)
c. Make sure to save color and depth sensor intrinsics, color to depth extrinsics and the depth-units (perhaps to JSON) for later
d. In ROS-bag recorder we are saving frames before synchronizing. Pipeline and syncer can possibly introduce some extra latencies. Check-out the [sensor-control](https://github.com/IntelRealSense/librealsense/tree/development/examples/sensor-control) example.
e. It would make sense to configure the color sensor to `YUYV` format instead of `RGB8`. This would save both on CPU and storage, without loosing information. 

Another consideration is making sure the hard-drive is actually fast enough to save all the data. Unfortunately, we can't integrate FFMPEG inside librealsense for RGB compression, and the bandwidth is still quite high for regular drives. A quick calculation brings me to ~240 MB/s for FHD color + HD depth @ 30 FPS, while non-SSD hard-drive speed can range anywhere from 50 to 300 MB/s as far as I know. So fast SSD would certainly help.

Hope any of this helps.

Vivikyung, 2018-10-18:
I'll try my best to get a faster SSD. In the meantime, is there a way to synchronize before using the ROS-bag recorder? Obviously I'm able to do it during playback, but perhaps it would reduce file size? Also, is there a link somewhere to help me with optimized compile? I'll try to do the rest of your suggestions as well and get back to you.

Vivikyung, 2018-10-29:
Hm. So if I don't display the images, I actually can record continuously in RGB format... So it was just continuously rendering the frames that caused my computer to freeze and crash.
Now I'm having the issue that when I play the RGB rosbag back, again continuously rendering the frames will freeze the computer. However, I'm going to close this issue because I can easily work around that situation by only displaying frames from the time periods I need.

closed 2018-10-29 by Vivikyung