realsenseai/librealsense #2564: Pointcloud `map_to` colorizer not saving properly in PLY

mprat, 2018-10-18:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version |   Linux 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-36-generic                                       | 
| SDK Version                            | 2.16.1                          | 
| Language                            |  python                     | 

### Issue Description

In version 2.16.0 and 2.16.1, the example code snippet for saving colorized pointclouds to PLY format does not save the correct color information.

```
import pyrealsense2 as rs

pc = rs.pointcloud()
pipe = rs.pipeline()
pipe.start()

try:
    frames = pipe.wait_for_frames()
    depth = frames.get_depth_frame()
    color = frames.get_color_frame()

    pc.map_to(color)
    points = pc.calculate(depth)

    print("Saving to 1.ply...")
    points.export_to_ply("1.ply", color)
    print("Done")
finally:
    pipe.stop()
```

When loading the PLY into MeshLab or CloudCompare, the RGB values are identical for all points in the pointcloud. When Bumping down to version 2.15.0, this bug disappears, and the RGB color is correctly saved in the PLY.

I have only tested this from the Python wrapper.

mprat, 2018-10-18:
I'm suspecting it has to do with this previous commit: https://github.com/IntelRealSense/librealsense/commit/b8fc6d4af645f87171f55db1183441fce9a85477

If I have time in a few days I can take a look. Do you guys accept community pull requests?

dorodnic, 2018-10-18:
Hi @mprat 
Did you had an opportunity to test from latest development branch?
I think this should be resolved, but we will double check prior to upcoming release.

mprat, 2018-10-18:
With this last commit from master: https://github.com/IntelRealSense/librealsense/commit/4c8fe6dd382fb7476fb3ce10cdf2eac1f264db14 (as of 5 minutes ago), and yes, this has been fixed.

When is the next release scheduled to come out?

dorodnic, 2018-10-19:
We are running late on 2.16.2, but should finalize it within couple of days.

eospi, 2018-10-28:
I've build the latest version, but the .ply file produced from export_ply_example.py still doesn't have the correct color values for the points. They all seem green/white. How can I resolve this? Thanks!

mprat, 2018-10-29:
Sounds like your color balance is off?

eospi, 2018-10-29:
Thanks for the reply! I've fixed it by waiting for some frames to go by before taking the point cloud.

HippoEug, 2018-10-29:
> Thanks for the reply! I've fixed it by waiting for some frames to go by before taking the point cloud.

#### C++ Equivalent 
```
for (auto i = 0; i < 10; ++i) { // Capture 10 frames for camera hardware to settle
		pipe->wait_for_frames();
}
```

closed 2018-10-30 by mprat