realsenseai/librealsense #2565: Realtime point cloud visualization in Python

ljc19800331, 2018-10-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello, I am wondering if there is build-in code for realtime point cloud visualization in Python for realsense? Thanks.

MartyG-RealSense, 2018-10-19:
If you prefer to generate the point cloud in Python within the SDK instead of using an external solution such as pyntcloud, the Python tutorial linked to below may be helpful to you.

https://github.com/dorodnic/binder_test/blob/master/pointcloud.ipynb

dorodnic, 2018-10-19:
Check out [Open3D](http://www.open3d.org/docs/tutorial/Basic/pointcloud.html)
We will add a tutorial soon for that as well

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi ljc19800331,

Did you try what MartyG-RealSense suggested? Still need any help?

Thanks!

ljc19800331, 2018-10-23:
Oh sorry for the late update. I have solved the problem with open3d. Thanks for your help.

This is an interesting question and I high recommend to have an example of realtime colorized point cloud visualization in python similar to https://github.com/IntelVCL/Open3D/issues/473#issuecomment-431506955 or https://github.com/toinsson/pyrealsense/blob/master/examples/show_vtk.py

closed 2018-10-23 by ljc19800331