realsenseai/librealsense #2569: CUDA support plan?

SamuelLordBlack, 2018-10-19:
In another thread, a contributor wrote "we are working on adding out-of-the-box CUDA support and planning to publish it soon". How soon is soon, and which components are planned? Specifically, I'm trying to decide whether I should write my own implementation of align or wait for something official.

We're currently running 2.16.1.

Thanks.

MartyG-RealSense, 2018-10-19:
After that mention of CUDA in June, CUDA optimized implementation was added in version 2.13.0 on June 22 2028.

https://github.com/IntelRealSense/librealsense/pull/1866

SamuelLordBlack, 2018-10-19:
That's basically two functions - format conversion and point clouds. The remainder of the library has not been touched, and that's what I'm looking for.

dorodnic, 2018-10-20:
Hi @SamuelLordBlack 
Image conversions and projection functions are a logical place to start, as they are most commonly used performance hits. It makes sense to add alignment next, and it is taking longer than I originally expected. We are constantly receiving feedback from our users, and trying to prioritize things based on that. The project also welcomes and relies on user pull-requests to better support non-Intel platforms, so if you do write your own implementation please consider sharing it. 
For the short term, our plans involve supporting more upcoming RealSense devices, continue investing in C# & Unity, expanding pypi distribution and Jupyter notebooks, build better integration with Open3D and OpenCV, improving CMake integration of samples and the Windows installer. In terms of optimizations my intuition was to invest in generic GPU acceleration next, but there is no official plan. Given more feedback from the Jetson community, we can go back to that instead.

SamuelLordBlack, 2018-10-22:
Thanks, Sergey. I'll see what I can drum up from the Jetson community. The align call currently takes 200ms on the Jetson TX2, so it's quite a bottleneck.

lesreaper, 2018-10-23:
Wow, that's a lot! I didn't think it was that big, though I did notice a slowdown on the TX2.

dorodnic, 2018-10-23:
I agree this requires a fix. We are looking into the issue (I'm a bit hesitant to commit, because align flow is somewhat more complicated than what we done before, and I'm not CUDA expert by any means. But we will do our best)

dorodnic, 2018-11-21:
CUDA optimized alignment is now part of librealsense 2.16.5.
@SamuelLordBlack thank you for bringing this to our attention. 
We still don't have a strict CUDA support plan, but we are committed to addressing problems that are important to our customers.

SamuelLordBlack, 2018-11-21:
Thanks! That's great turnaround time. We're looking forward to pulling that down.

RealSense-Customer-Engineering, 2018-12-21:
[Realsense Customer Engineering Team Comment]
Hi SamuelLordBlack,

If not yet, you may subscribe to RealSense SDK info and check back for CUDA related improvement.

If nothing else is needed, we will close this one soon.

Thank you!

closed 2018-12-27 by dorodnic