realsenseai/librealsense #2570: How to display edge, shape of an object or surface of an object

NguyenThaiHoc1, 2018-10-20:
Dear Technique-Realsense support,
I have some question about technique for developing on intel D435 Depth sensor 

First question, Can I catch all shape of an object  ? and How can i do ? (Point cloud ? or something) 

Second question,  How to display edge, shape of an object or surface of an object 

Final question, Can i generate an object which i catch image? 

I'm Python Developer (if you have the code better)

Sorry for my English skill is bad
Thanks you for support 
NguyenThaiHoc1,

MartyG-RealSense, 2018-10-20:
**Question 1**

You can capture an object as a point cloud.  To capture all 360 degrees of the object though, you will have to do one of these techniques:

- Use multiple cameras to capture point clouds from different viewpoints and then combine the point clouds together into a singke point cloud.

- Move a single camera around the object.  Or

- Rotate the object 360 degrees using a method such as a turntable, or attaching the object to a rotating robot arm.

**Question 2**

The 400 Series cameras support post-processing processes to apply a range of filters to data, such as edge-preserving.  

A good place to start is Intel's post-processing white paper document.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-Depth-PostProcess.pdf

Intel has also published a Python tutorial for applying post-processing filters.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

**Question 3**

Do you mean that you want to create a solid 3D 'mesh' model of an object?  If that is what you mean then yes, you can convert a point cloud into a 3D object.  This can be done using 3D model software such as MeshLab or Blender.

There is also a program called LIPScan that works with the 400 Series and can capture a solid model directly.

https://www.youtube.com/watch?v=N14Pi6z-MkE

NguyenThaiHoc1, 2018-10-21:
In question 1, you can give me a example code or something like ???? software and how to use ??? 

software which you present in link above https://www.youtube.com/watch?v=N14Pi6z-MkE (what do it called ? and how i dowloand it ?)

Question 2: what do you mean - filters (or post-processing filters.) ? Why do we need ? 

Question 3: I try to using save file .ply (point cloud) from Intel-realSense-Viewer application. But i loading this file on MeshLab. it very bad and just display some a some area black ? 

Sorry for my technique is not good Because I'm newbie in 3D-technology
if you have a some document for newbie learn 3D-technology, you can send to me on this issue
Thanks you for support

MartyG-RealSense, 2018-10-21:
This link provides an introduction to stereo vision technology on Intel's blog, written by RealSense SDK Manager Sergey 'Dorodnic' Dorodnicov.

https://realsense.intel.com/stereo-depth-vision-basics/

**Question 1**

There is very little information available about using LIPScan with the 400 Series other than the YouTube video, as the LIPScan website mostly focuses on using LIPScan software with LIPScan's own depth camera.  

My assumption from the YouTube video is that the LIPScan software may work with the 400 Series depth camera too, as the LIPScan software shown on their website looks the same as the one being used with the D415 in the YouTube video

On the page linked to below, click the More button and scroll down the page to find the software and user manual links.

https://www.lips-hci.com/product?product_id=16

**Question 2**

Filters are software functions that apply changes to camera data to alter it in some way that improves the results, such as reducing holes in the image.

**Question 3**

Would it be possible please to post a picture of your MeshLab import results here on this discussion so we can see them?

NguyenThaiHoc1, 2018-10-22:
Hi @MartyG-RealSense 
I have some picture you can see 

Picture i save ply from app Intel-RealSense-Viewer
[image]

Picture when i loading ply file on MeshLab. it is terrible
[image]

MartyG-RealSense, 2018-10-22:
For some reason, your reply appeared in my email box but not on this discussion.  so I will post the pictures that you kindly sent of your ply that you made in the Viewer and imported into MeshLab so that others can see them.

[image]

[image]

There is a guide to importing a ply into MeshLab.  Looking at its images, it does seem that the ply looks worse after import than it did originally in the Viewer, but that the final 3D model turns out okay once you have applied the settings in the guide.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

NguyenThaiHoc1, 2018-10-22:
hi @MartyG-RealSense 

I really appreciate your help with my project. But I realize that i need such RGB-D Scenes 

How to I can create RGB-D sences like RGB-D washington Dataset (link below)  ? 

Washington dataset: https://rgbd-dataset.cs.washington.edu/

I'm doing my project but I have some issue thinking about hardware camera 

Structure I practice, I have a problem about Depth dataset - Depth CNN (this is main reason I buy Intel D400 series) I hope i can create a lot of datasets which apply on this Model
[image]

MartyG-RealSense, 2018-10-23:
You are very welcome.  I'm glad to be able to be of help.

The 400 Series cameras use Stereoscopic vision, with a left and right imager.  Kinect 1 was based on Structured Light, which is similar to the RealSense SR300 camera model.  Kinect 2 used a different system called Time of Flight.

Dorodnic the RealSense SDK Manager published a blog article explaining how stereo depth vision works.

https://realsense.intel.com/stereo-depth-vision-basics

I do not have any knowledge of using DNN though, so hopefully one of the Intel guys on this forum can give you the advice you need about that.  Good luck!

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Hi NguyenThaiHoc1,
If there is no more question about this topic, we'd like to close this ticket. Thanks!

closed 2018-12-10 by RealSense-Customer-Engineering