realsenseai/librealsense #2572: D415 not recognised in Windows 10, Surface Pro 4

erget, 2018-10-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   {Win 10 / Also tried with Ubuntu 18 | 
| Kernel Version (Linux Only)    |  Not applicable                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.1.251 }                          | 
| Language                            |  Target: Python                          | 
| Segment			|  Others                          | 

### Issue Description
Windows doesn't recognise my RealSense D415. I'm using a Surface Pro 4 with the camera connected to the USB 3.0 port and while the Device Manager reports no "Imaging devices" at all, although the RealSense is discovered under "Universal Serial Bus devices" (see attached screenshot).

I've updated the firmware and streamed RGB and depth from Ubuntu using the `realsense-viewer` without problems. Do you know what's causing the problem in Windows? Thanks! :)

[image]

MartyG-RealSense, 2018-10-21:
Historically, RealSense cameras have experienced problems in the past with Surface Books and Surface Pro 4's, apparently related to the USB ports.  The problems seem random - some owners have had problems whilst others haven't.   

I note though that you have been able to use the camera successfully in Ubuntu, suggesting that if you were running Ubuntu on the Surface Pro 4, in this case it is more a Windows issue than a general USB issue.

 Could you please try opening the Settings window of Windows.  you can find this by typing Settings into the text box at the bottom of your Windows screen, next to the button where you shut the PC down.  On the Settings window, please go to the Privacy option, then the Camera option, and check if 'Allow access to the camera on this device' is set to On.

[image]

erget, 2018-10-21:
Hi @MartyG-RealSense thanks for the quick response! Actually I have a dedicated Windows box and although it runs WSL, I had been accessing the camera from Ubuntu running on another machine. So you might be right about it being a USB issue.

I checked my settings and apps have access to my camera, so that doesn't seem to be the problem.

When you say that the USB access problems seem to be random, is there any kind of diagnostic I could perform to find out if I should expect the RealSense to run on my Surface? One of the parameters of the project I'm working on is that it should be a tablet, so I'd really like to be able to develop it on the Surface if possible.

MartyG-RealSense, 2018-10-21:
RealSense 400 Series cameras have the advantage of being able to work with any Intel or ARM processor.  So problems that occur tend to be related to the USB or the computer's operating system.

The SDK has a debug tool called 'rs-enumerare-devices' that can be used to provide information about the camera in a console window.  It can be found pre-built in the RealSense SDK 2.0 > Tools folder of the SDK.

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

MartyG-RealSense, 2018-10-21:
You could also try the camera in a USB 2.0 port on the Surface Pro 4 if you do not have a second USB 3 port to try the camera in.  The 400 Series camera can work in a USB2 port, though with lesser capabilities.  If it works in a USB 2.0 port, it would strengthen the evidence for the problem being related to the USB 3.0 port.

erget, 2018-10-21:
OK, thanks for the tip. Unfortunately the troubleshooter wasn't able to find any problems... I'll keep looking.

MartyG-RealSense, 2018-10-21:
The complicating factor with the Surface Pro 4 is that it only has a single USB 3.0 port and no USB 2.0 ports unless you use the device with a dock.  So a USB problem might typically be corrected by using a mains-powered USB port with the camera instead of plugging the camera directly into the computer.  However, a mains-powered hub would tether the computer to the proximity of the mains plug, greatly reducing portability.

erget, 2018-10-21:
Alright. Maybe I'll give it a shot with a hub tomorrow then. Thanks!

RealSense-Customer-Engineering, 2018-10-24:
[Realsense Customer Engineering Team Comment]
@Erget Did you get chance to try with hub? And do you still have the issue? Looking forward to your update. Thanks!

erget, 2018-10-25:
Sorry, I haven't had a chance to obtain a hub yet, and I'll be out of town on the weekend. I hope that I get a chance to take a look next week. Sorry for moving so slow on this!

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
@Erget Any good news with hub? Looking forward to your update.

erget, 2018-11-08:
Hi, sorry, today we made the decision to use a ZED camera instead (not really related to the hub issue, but it's one of the reasons this was pending for so long). That means that this issue will be mothballed from our side at least for a while. Sorry about the long wait and thanks for the patience.

closed 2018-11-08 by erget