realsenseai/librealsense #2573: Where is the source code for the alignment function in RealSense SDK2.0?

njnydxzc, 2018-10-21:
Hello,
 I am trying to figure out where the source code (C++) is for the alignment function in RealSense SDK2.0 and wonder if someone can give me an advice. 
Specifically, I wrote codes for aligning a depth frame to the corresponding color frame as follows:
 
rs2::frameset frames = pipe.wait_for_frames();
rs2::align align(rs2_stream::RS2_STREAM_COLOR); 
rs2::frameset aligned_frames = align.process(frames);  
rs2::frame color_frame = frames.get_color_frame(); 
rs2::frame depth_frame = aligned_frames.get_depth_frame();
  
I'd like to see source codes for checking how the alignment is accomplished. I found the following codes in "rs_processing.hpp" but I do not see any code for actual calculations for the alignment.

class align
{
public:
.....
}
  
Also, I've found that "align.cpp" has the "void align_images()" function but I am not sure if this is the same as the alignment function above.
Where can I find the source code for the alignment function?

MartyG-RealSense, 2018-10-21:
This same question was discussed on the Intel Support website in June 2018.  An investigation found the same information that you did.  

I recommended that the user in that case contact an excellent RealSense user called UnaNancyOwen, who has developed great OpenCV-based samples for SDK 2.0 and also provides the best unofficial documentation site for the SDK.  The user said that UnaNancyOwen replied that the alignment function is written in the align.cpp file.

A member of the Intel team on this forum will likely be able to provide a better official answer (I am not an Intel employee).

dorodnic, 2018-10-22:
Everything is inside [align.cpp](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp). That said, the file currently contains two implementations - one generic C++ and another optimised for Intel platforms.

njnydxzc, 2018-10-28:
I got RGB and depth images separately through the Intel RealSense Viewer tool. But these two pictures do not correspond. I got a coordinate by image recognition in the RGB image. How do I get the corresponding coordinates of the coordinates in the depth image to get the depth value corresponding to it?

HippoEug, 2018-10-28:
[image]

You want the two "previews" as seen above, color and depth to be aligned correct? I too am trying to find a solution for this.

I think the first step is to match the resolution of the streams.. the default for color is something like 640x480 while the depth is 1280x720.

dorodnic, 2018-10-28:
This is somewhat repeating question - it is impossible to align images based on image data alone. You need camera calibration information. You could write elaborate code to serialize and deserialize this data yourself, but we strongly recommend using the built-in recorder ([example](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback))

We do plan to provide an example of how the ROS-bag recorder can be used to record just a single frames pair (based on popular demand), but it can take some time until its ready to publish.

agrunnet, 2018-10-28:
Also, you are too close to the object. The "starry sky" in the lower part of the image is an indication that you are with the minZ operating range. Please move farther away, or reduce the resolution (which will reduce the minZ), or use the disparity shift. All are described in the white papers on the RealSense website.  https://realsense.intel.com/intel-realsense-downloads/#whitepaper

HippoEug, 2018-10-29:
> Also, you are too close to the object. The "starry sky" in the lower part of the image is an indication that you are with the minZ operating range

Yeap I understand that. Didn't bother to pick the camera up as it was resting on the table! 👍 

[image] [1280x720]

Here is a better example of what @njnydxzc is saying, probably that the Depth image has a wider FOV than the Color image, hence the (x, y) coordinates do not correspond.

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
Hi njnydxzc,

With what dorodnic has provided, do you still need any help?

Thanks!

njnydxzc, 2018-10-31:
If I want to manually complete the correspondence between RGB and depth pictures and find the corresponding coordinates, how can I accomplish this with camera calibration information?Which method should I use to make them correspond?

HippoEug, 2018-11-01:
Thank you @njnydxzc for asking. I also have the same questions, and have not looked into it. Any solutions to this question would help us intensely.

closed 2018-12-12 by RealSense-Customer-Engineering