realsenseai/librealsense #2575: Received Automatic Firmware Downgrade: 5.9.2 to 5.8.15

xtrawurst, 2018-10-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.2 | 
| Operating System & Version |   Win10  | 

### Issue Description
Connecting my D415 to my computer I just received a firmware "update" to 5.8.15 even though the camera was already running 5.9.2 (I put that on manually using the Intel DFU tool).

Is this expected behavior?

dorodnic, 2018-10-23:
No... In my opinion this should be considered a bug. Can someone from the support team please assist to reproduce? @xtrawurst it would help us if you could share the exact Windows build you are using? Anything odd about it?

danielw256, 2018-10-23:
https://github.com/IntelRealSense/librealsense/issues/1996

xtrawurst, 2018-10-23:
@dorodnic 
Windows 10 Pro, Version 1803 (OS Build 17134.345).
This is running on a MacBook Air using a separate partition (using BootCamp).

RealSense-Customer-Engineering, 2018-10-24:
[Realsense Customer Engineering Team Comment]
Tried D415 (FW5.9.14) on NUC6i3SYH with Windows 10 Pro (version 1803 OS build 17134.345), but couldn't reproduce the issue even with the "RS400 Camera Service" enabled. The camera driver version is 5.1.160.24 (6/12/2018)

Thanks

RealSense-Customer-Engineering, 2018-11-09:
[Realsense Customer Engineering Team Comment]
Is this issue still there after disabling "RealSense Camera DFU Service" and "RS400Service" in Windows 10?

xtrawurst, 2018-11-12:
The RS400Service is running at the moment on the target machine. However this is after disabling it before. It keeps coming back and there seems to be no way to turning it off for good.

RealSense-Customer-Engineering, 2018-11-12:
[Realsense Customer Engineering Team Comment]
It sounds interesting. I remembered when I had same issue before, I just simply fixed it by disabling the Service.

Can you please try this:
1) Search the binary file "RS400Firmware.bin" in Windows (\system32\...) , rename it to whatever different one. And try again....
or
2) Reinstall your Windows or try different computers...

Thanks

RealSense-Customer-Engineering, 2018-11-19:
[Realsense Customer Engineering Team Comment]
Hi
Have you tried or is this issue solved?

xtrawurst, 2018-11-19:
@RealSense-Customer-Engineering 
Not yet. I'll try soon.

closed 2018-12-03 by RealSense-Customer-Engineering