realsenseai/librealsense #2576: .bag file open in Matlab

LouMagsClark, 2018-10-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2                          | 
| Language                            |  Matlab                          | 
| Segment			|  Robot                  | 

I'm trying to open the recorded .bag file in Matlab using the Robotics System Toolbox, and loading the rosbag file as a structure, although this isn't working and I'm not sure that structure is necessarily what I need. I haven't found any suggested codes on the Mathworks pages which have opened the RealSense .bag files. 

I am able to open the 3D snaphot PLY files in Matlab, but would like to take a recording of the area to gain a better fill % (I'm at a working distance of 230 mm, using the High Accuracy preset and a disparity shift to account for the MinZ).

Any help with this is greatly appreciated, sorry if this is too specific for this forum!
Cheers

LouMagsClark, 2018-10-23:
After coming back to the issue today, I guess what I would really like to do is stream from the D415 and save the captured stream within an x,y,z axis, like a .PLY file

I'm currently streaming for 5 seconds using this code I adapted:

`function stream_example() 

    %Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    %Make Colorizer object to prettify depth output
    colorizer = realsense.colorizer();
    %Start streaming on an arbitrary camera with default settings
    profile = pipe.start();

    tic
        while toc < 5

        fs = pipe.wait_for_frames();
        depth = fs.get_depth_frame();
        dist = depth.get_data();
               
        color = colorizer.colorize(depth);
        data = color.get_data();
        img = permute(reshape(data',[3,color.get_width(),color.get_height()]),[3 2 1]);
        imshow(img);
        
        end
    
end`

but I would like to save the totality of the stream, and view it in a 3D format, rather than the current imgshow 2D style.

Angain, any help greatly appreciated! I'll keep working on this too and report back with any updates from my end.

Cheers again

lramati, 2018-10-29:
You can use the realsense.pointcloud class to convert depth data into 3d data (This can also texture map supplied color frames). The resulting output will give you a list of x,y,z coordinates with corresponding u,v texture coordinates if you supplied a color frame. However, I'm not sure how to display this visually with built in matlab functions. Maybe somebody more familiar with matlab can help with that part?

HippoEug, 2018-10-30:
> You can use the realsense.pointcloud class to convert depth data into 3d data (This can also texture map supplied color frames). The resulting output will give you a list of x,y,z coordinates with corresponding u,v texture coordinates if you supplied a color frame.

Hey, sorry to hijack the thread. Is the pointcloud class mentioned a C++ or Python example? If it is in C++, do you mean [this, rs-pointcloud.cpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp)? Many thanks.. I am interested in the x, y, z data.

lramati, 2018-10-30:
That's exactly what I was referring to

dorodnic, 2018-10-31:
@ev-mp can you share the example you've been working on?

HippoEug, 2018-11-01:
I have asked this particular part of the question in a separate issue [here](https://github.com/IntelRealSense/librealsense/issues/2584#issuecomment-434147781).
But for the sake of readability I'll write it here as well:

#### Regarding the `rs-pointcloud.cpp`:
Since the `rs-pointcloud.cpp` or pointclouds in general are made up of (x, y, z) coordinates, why can't we just extract the (x, y, z) coordinates from the function which draws the pointcloud in `rs-pointcloud.cpp`?

#### `Here is a snippet of the pointcloud code from example.hpp:`
```
/* this segment actually prints the pointcloud */
auto vertices = points.get_vertices();              // get vertices
auto tex_coords = points.get_texture_coordinates(); // and texture coordinate
for (int i = 0; i < points.size(); i++) // this loops 921,600 times, so every single pixel. The data should be here
{
    if (vertices[i].z)
    {
        // upload the point and texture coordinates only for points we have depth data for
	    glVertex3fv(vertices[i]);
        glTexCoord2fv(tex_coords[i]);
    }
}
```
The problem I have currently is that I am unable to get any (x, y, z) coodinates from the above function, just do not know where the code is "hiding"!

I do not believe the particular problem listed here requires Matlab wrapper.. fill me in if I am wrong please. Thank you all very much and I look forward to some suggestions!

LouMagsClark, 2018-11-02:
Within the RealSense Viewer Program you're able to take snapshots in 2D (.png), export a snapshot of the 3D model (.ply). Using the Matlab wrapper I can stream the 2D image, however I can't see how I can stream and save the 3D model version. 

As you say, it may not be wrapper based and I may be missing a trick with converting the 2D to 3D (I'm working on finding out if that is the case).

lramati, 2018-11-07:
As I mentioned above, using the [realsense.pointcould](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/pointcloud.m) class, you can feed in depth frames (and optionally color frames) to get out [points](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/points.m) objects. This is a type of frame that exposes and additional function called get_vertices, which return an N-by-3 matrix containing the x,y,z coordinates of each pixel in the original depth frame.
This can be done on each frame in real time, however I personally don't know enough matlab to be able to help you visualize the data.

LouMagsClark, 2018-11-07:
Thank you for your help @lramati, sorry for not acknowledging your earlier post, I'll report back once I have tried this out.

closed 2019-02-04 by RealSenseCustomerSupport

TDVL, 2019-02-23:
Dear LouMagsClark and RealsenseCustomerSupport,
I saw your post about RealSense and Matlab.
I have a RealSense D400 series cameras, I am using RealSense Viewer in Win10, the Viewer stores data in bag file.
How can I extract the point cloud from bag file in Matlab. I can access the RGB and INFRA data, but I can't access the point cloud's XYZ coordinates array?
Also, the Matlab wrapper provided by Intel (pointcolud_example function) did not help me.
Can you help me about this?
Do you have a code for Matlab for point cloud extraction with RGB and XYZ arrays?
Also, when I use the Viewer, I can export the scene in PLY format, but the resolution in PLY format is not preserved. I need a 640x480x3 (307200x3 XYZ and RGB arrays). Is it possible to set somehow?
Thank You in advance.
Kind regards

QuocZung, 2019-04-03:
Hi @RealSense-Customer-Engineering . I also want to do something similar to this issue. I am able to stream Color, Depth, Aligner frames and get out all pointcloud vertices but have problem to render it in C#. So is there any possible method powerful enough to render the pointcloud with C# which allows you to zoom, rotate the 3D view ,etc. or simpler like save the stream 3D view as 3D model (maybe .ply) file?