realsenseai/librealsense #2577: Applying a preset in pyrealsense2

dibya-pati, 2018-10-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi @dibya-pati,

Try below code to see how it works:

    pipe_profile = pipeline.start(config)

    depth_sensor = pipe_profile.get_device().first_depth_sensor()

    preset_range = depth_sensor.get_option_range(rs.option.visual_preset)
    _#print('preset range:'+str(preset_range))_
    **for** i **in** range(int(preset_range.max)):
     visulpreset = depth_sensor.get_option_value_description(rs.option.visual_preset,i)
     print(**'%02d: %s'**%(i,visulpreset))
     **if** visulpreset == **"High Accuracy"**:
     depth_sensor.set_option(rs.option.visual_preset, i)

dibya-pati, 2018-10-23:
> [Realsense Customer Engineering Team Comment]
> Hi @dibya-pati,
>
> Try below code to see how it works:
>
> ```
> pipe_profile = pipeline.start(config)
>
> depth_sensor = [...]

Thanks, this works

closed 2018-10-24 by RealSense-Customer-Engineering

lxop, 2022-11-28:
Hi @RealSense-Customer-Engineering ,

Is there any intention to expose the preset symbols in the Python wrapper directly so that we can do something more like
```python
depth_sensor.set_option(rs.option.visual_preset, rs.rs400_visual_preset.high_accuracy)
```
rather than looping through descriptions looking for a label just to set a preset? Or is this looping approach intended to be the way to apply a preset with the Python wrapper?

It seems like it would simply require one line in `c_files.cpp` to do this - is it being avoided intentionally? I'm happy to open a PR if that's all it requires.

Thank you,
Alex

MartyG-RealSense, 2022-11-29:
Hi @dibya-pati  You should be able to set the preset with a simple number value if you know which number relates to which preset on your particular camera (the 400 Series and L515 models have different presets).  A Python example of this simplified code would be:

`depth_sensor.set_option(rs.option.visual_preset , 3)`

On 400 Series cameras, presets start at '0' and correspond to the order in which presets are listed in the RealSense Viewer.  So if you open the Viewer's preset list and count 0, 1, 2, 3, then '3' would be the fourth preset in the list, which is 'High Accuracy'.

[image]

lxop, 2022-11-29:
Hi @MartyG-RealSense , 

Thanks for your reply; I understand that I can use a plain number like this, however it would be valuable to have these values exposed symbolically in the library, so I'm just wondering if they have been omitted intentionally or just haven't been done yet.

Thanks,
Alex