realsenseai/librealsense #2578: Setting max distance via Python

lesreaper, 2018-10-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.10.06 | 
| Operating System & Version |   Jetson TX2  | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                      | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            | 2 }                          | 
| Language                            |  {python}                          | 

### Issue Description
I'm trying to access and set the max_distance on the camera via Python.  Here's the code, which I pulled from the IDE recommendation on the functional call.

```
pipeline = rs.pipeline()
config = rs.config()
profile = pipeline.start(config)
depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(rs.option.max_distance, 16)```

I get this error, which is odd for a depth sensor: 
RuntimeError: Device does not support option Max Distance!

ev-mp, 2018-10-23:
Hello @lesreaper ,
The specific options are applicable with colorizer object.
To check whether specific option is available use the `supports` API
> depth_sensor = profile.get_device().first_depth_sensor()
   if (depth_sensor.supports(rs.option.max_distance)):
        ...

RealSense-Customer-Engineering, 2018-10-24:
[Realsense Customer Engineering Team Comment]
Hi @lesreaper,

Try below code to see how it works.

colorMap = rs.colorizer()
**print**(**"colorizer - max_distance: %f"** %(colorMap.get_option(rs.option.max_distance)))

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
Hi @lesreaper,

Still need any support for this topic?

closed 2018-11-19 by RealSense-Customer-Engineering