realsenseai/librealsense #2581: Tell if connected to D415 or D435

mprat, 2018-10-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.14.13                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  2.16.1                          | 
| Language                            |  C++ / Python                          | 

### Issue Description

I am using the Python and/or C++ APIs to talk to Realsense cameras. How can I tell if I am connected to a D415 vs. a D435? They take slightly different parameters, and I want to start a slightly different set of streams for both types of cameras. What is an easy (programmatic, through the API) way to tell which type of camera I am connected to?

I can upgrade firmware, update my library version, etc. if this is necessary to make this happen.

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi @mprat,

You can refer to the sample code "python-rs400-advanced-mode-disabled.py" to check the PID which is different between D415 (0AD3) and D435 (0B07).

mprat, 2018-10-23:
Awesome, thanks!

closed 2018-10-23 by mprat