realsenseai/librealsense #2582: Announcement: Dot3D Scan software now compatible with D410, D415 and D435

MartyG-RealSense, 2018-10-23:
Hi everyone,

I just wanted to give a heads-up that DotProduct's Dot3D Scan 3D scanning software is now compatible with RealSense D410, D415 and D435.  

https://communities.intel.com/thread/130133

This software will also be compatible with DT's new military grade DT301-X RealSense equipped tablet coming in November 2018.

https://www.officer.com/command-hq/technology/computers-software/tablets-handhelds/press-release/21027456/dt-research-the-dt301x-rugged-tablet-is-meant-for-high-accuracy-measurements

LouMagsClark, 2018-10-24:
I downloaded a free trial of the software, it crashes every time I select new scan... This is with a D415 camera.

MartyG-RealSense, 2018-10-24:
DotScan 3D's hardware specification requires a USB 3.0 port.  Are you using your D415 with USB 3, please?

LouMagsClark, 2018-10-24:
Yes, with a USB 3.2

MartyG-RealSense, 2018-10-24:
And you have a computer with an Intel Core i5 processor or faster, and graphics that support OpenGL 3.3 or higher?  These are the two other specification recommendations of the software.

dorodnic, 2018-11-01:
Since there is no clear closure policy for announcements, lets agree they will be closed once they leave the front page.
Thank you, @MartyG-RealSense! Your strong involvement in the community is highly appreciated.

closed 2018-11-01 by dorodnic

MartyG-RealSense, 2018-11-01:
Yes, that's a very fair policy.  I'm aware I'm adding to the count of open cases with such announcements, even though they are very occasional.  So I totally agree with that policy  Thanks for your understanding, and your kind words.  *smiles*