realsenseai/librealsense #2585: Short Range Depth & Discontinuous Video

philipjames44, 2018-10-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300| 
| Operating System & Version |  MacOS  | 
| Platform                                 | PC |
| Language                            |  C++                        | 

I was wondering if there might be a built in method to set my camera to short-range depth detection.  Not using the GUI, but using the same values that the GUI uses for its short-range preset? If not, what's the best way of going about this?

Unrelated to the first part, but I am recording videos in segments. Between segments, I have approximately 3 seconds of video that are not recorded. My code is as so for the file switch:

` pipe->stop();
cfg.enable_record_to_file(filename);
pipe->start(cfg); //File will be opened at this point
device = pipe->get_active_profile().get_device();`

Obviously the pipe stops and starts which causes the loss of a few seconds. But would there be anyway for the camera to not reset, but rather record continuously through a pipeline, and simply change output files at desired time increments?

philipjames44, 2018-10-24:
So for part II with altering depth with presets, this was wonderful: https://github.com/IntelRealSense/librealsense/issues/2193

However I am still having issues trying to figure out how to record consecutive video sequences in separate files without losing a couple seconds of footage. Any help would be appreciated :)

HippoEug, 2018-10-24:
Ha! I am trying to do the opposite of you, I am trying to capture only 1 frame, not even 1 second!

philipjames44, 2018-10-24:
If you only want 1 frame you can follow this tutorial: https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
Hi @philipjames44,

What's your cfg (resolution, fps, etc) for depth and color? From the frame number indication, I can see some frame drops in the beginning of the streams on my setup, but doesn't observe a few seconds of frame drops. between pipe stop and start.
Does this happens on your Mac or your also try it on another system?

philipjames44, 2018-10-25:
Hi @RealSense-Customer-Engineering ,

At the moment I only have a Mac to test on, however I will test on an Ubuntu system in approximately a week. My depth and color are both 640x480 for resolution at 30fps. The thing is, my Mac is rather powerful, so I can't imagine my system is the bottleneck. My specs are an intel core i7 at 2.6 GHz, 32GB of DDR4 ram, and 2TB of SSD storage. Even when I turn all other apps off, the delay doesn't change. The delay is also constant per duration. Ie, recording in 1 minute increments and recording in 15 minute increments both produce a constant, nearly exact, 3 second delay.

RealSense-Customer-Engineering, 2018-11-02:
[Realsense Customer Engineering Team Comment]
Hi @philipjames44,

To test based on librealsense 2.16.3 with VGA resolution and 30FPS for depth and color, the time diff between pipe->stop() and the 1st frame receiving time I observed on one Mac Air is about 100ms. SR300 firmware version is 3.21.0.0.
You can also enable the librealsense log file to show the result.

philipjames44, 2018-11-02:
@RealSense-Customer-Engineering 

I believe I am using the same SDK version (whatever is the most recent non-experimental build is what I'm using). I can confirm the time delay is at least 3 seconds on my machine. However I have never manually updated the camera firmware. Is this automatic?

RealSense-Customer-Engineering, 2018-11-07:
[Realsense Customer Engineering Team Comment]
Hi @philipjames44,

You can use realsense-viewer or rs-enumerate-devices app to check the firmware version first.
Refer to below article and install the DCM driver which contains SR300 latest firmware (3.21.0.0) on Win10 with your SR300 connected. The DCM will upgrade the firmware automatically.
https://www.intel.com/content/www/us/en/support/articles/000006224/emerging-technologies.html

philipjames44, 2018-11-23:
@RealSense-Customer-Engineering The firmware is the most recent version `Firmware Version              : 	3.21.0.0`

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

Hi @philipjames44,

Are you able to reproduce this issue in the latest librealsense on your platform, either Mac or Ubuntu? I still have trouble to reproduce this.

RealSenseCustomerSupport, 2019-06-18:
----------------------------------------------

Hi @philipjames44,

Still need any support for this topic? If not, will close this.

philipjames44, 2019-07-07:
If anyone lands on this issue I never found a solution. The issue was consistent across Ubuntu and Mac OSX. I just lived with a few seconds of lost footage.

closed 2019-07-07 by philipjames44