realsenseai/librealsense #2586: Could not find a version that satisfies the requirement pyrealsense2

kqi914, 2018-10-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Operating System & Version |    Linux Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                    | 
| Platform                                 | NVIDIA Jetson TX2 L4T 28.2.1 |
| Language                            |  python                       | 
| Segment			|  Robot                    | 

### Issue Description
When trying to download pyrealsense2 using pip2 i get the following message: 

```
nvidia@tegra-ubuntu:~$ pip2 install pyrealsense2
Collecting pyrealsense2
Could not find a version that satisfies the requirement pyrealsense2 (from versions: )
No matching distribution found for pyrealsense2
You are using pip version 8.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

pip3 and pip give out this message: 
```
Traceback (most recent call last):
 File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
```

I have looked everywhere and I found a similar post for MAC OS but it is not helpful for me. Also, I tried uninstalling and reinstalling pip and pip3 but the issue doesnt seem to go away. Any clues as to how to solve this issue or how to install the pyrealsense2 module manually?

Note: I have been using Ubuntu for about 6 months so I am not very good with it but I think pip and pip3 are not working because I upgraded them using the command above, hence the use of pip2.  See #5221

dorodnic, 2018-10-24:
There is nothing wrong on your side. Unfortunately, we still not ready to publish pip packages for ARM architecture (and same goes for Mac, although this one should be solved soon)
For the moment, if you want to work with `pyrealsense2` on the Jetson, you need to [build it from source](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source). Overall, it is not very complicated and should take about half-an-hour.

kqi914, 2018-10-25:
> There is nothing wrong on your side. Unfortunately, we still not ready to publish pip packages for ARM architecture (and same goes for Mac, although this one should be solved soon)
> For the moment, [...]

@dorodnic On step 3 where it says:
`cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
`
Is that command by itself supposed to work or should i add something to it? I am getting the following message:

```
nvidia@tegra-ubuntu:~/build$ ../ -DBUILD_PYTHON_BINDINGS=bool:true
CMake Error: The source directory "/home/nvidia" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```
I tried adding the path to the downloaded files for pyrealsense2 but it gave me the same message. I have never built anything from source so this is pretty new to me, hope you can help me with this.

dorodnic, 2018-10-25:
Of course, seems like just a mix-up with the current directory. Try:
```
cd ~
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir build
cd build
cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
make
```
I do want us to expand the `apt` packages beyond Intel-only platforms, but it can take some time. Sorry for the inconvenience.

RealSense-Customer-Engineering, 2018-10-26:
[Realsense Customer Engineering Team Comment]
Hi kqi914,

Besides what have been mentioned by dorodnic, here is what I used to build from LibRealSense source code off Nivida TX2:

As there could be several versions of python installed, I wanted to specify the executable path of the python version to be used (I used python 2.7 here)

mkdir -p ~/rs_2.16.1
cd ~/rs_2.16.1
git clone https://github.com/IntelRealSense/librealsense.git .
mkdir -p build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python2.7
make -j4 && make install

Hope this can help clarify.

Thanks!

kqi914, 2018-10-30:
After having some issues with Cmake I made it work using the steps provided by dorodnic. However,  in the last step:

> Alternatively, copy the build output (librealsense2.so and pyrealsense2.so) next to your script.

I could only found the file named "librealsense2.so", I could not find the other one although there was one with a similar name inside the wrappers folder called "pyrealsense2.cpython-35m-aarch64-linux-gnu.so". 

I went ahead and ran the program sometimes it runs, other times it says "the device is not connected5000" even when it is, but a change of USB port will fix this issue. 

One last thing, if I close the terminal and reopen it I have to run the command
`export PYTHONPATH=$PYTHONPATH:/usr/local/lib
`
again for the python script to be able to use the pyrealsense2 module. is there a fix around this?
Thanks!!

kqi914, 2018-11-06:
I still need to run: `export PYTHONPATH=$PYTHONPATH:/usr/local/lib ` every time I want to run the python script. Any solution for that?

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Hi kqi914,

There are different ways to take care of python env settings. One of the things that you may try is adding it to .bashrc file.

Thanks!

RealSense-Customer-Engineering, 2018-11-12:
[Realsense Customer Engineering Team Comment]
Hi kqi914,

I am closing this one now, and please feel free to report back any issue or ask questions.

Thanks!

closed 2018-11-12 by RealSense-Customer-Engineering

iggy12345, 2019-06-14:
I don't understand, this literally worked earlier today and now it doesn't, why else would it be saying this?

I'm on windows 10 using python 3.6 with mingw installed

lramati, 2019-06-16:
Please open a new issue with a detailed explanation of your problem

malith1992, 2019-06-25:
I try to build the pyrealsense2 from source. But I can't understand what the following step mean
"Alternatively, copy the build output (librealsense2.so and pyrealsense2.so) next to your script." . I just copied the two files in to usr/local/lib but I think that is not what it want me to do. 

I get the "No module named 'pyrealsense2' error when I try to import pyrealsense2 in my python code.

lramati, 2019-08-20:
usr/local/lib is not the right place to put these files. That is the path linux uses when trying to run programs, what you need to do is put librealsense2.so and pyrealsense2.so on the pythonpath, that is, where python looks for modules when importing.
You can check where your python install is looking for modules by running
```python
import sys
print(sys.path)
```
See [this Stack Overflow question](https://stackoverflow.com/questions/31384639/what-is-pythons-site-packages-directory) for more details

gavan1, 2020-02-20:
@lramati @malith1992 it seems that there still isn't any full step by step documentation on this process... Truly surprises me as  I was under the impress INTEL wants to have adoption of their tools.

Python will be very new for a lot of of people one simply because of the new found demand in AI and Computer vision for it.

I have the 2.32.1 SDK correctly installed on my 
Jetson TX2

cMake examples compile and run.

Since it's the arm architecture python modules need to be build from source.
once built via cmake, are found ..
` librealsense/build/wrappers/python`

pybackend2.cpython-36m-aarch64-linux-gnu.so.2.32.1
pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.32.1

current official live documentation says to 
`sudo make install`

On a Jetson TX2 this moves the the files into  `usr/local/lib`
I have also run
`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`
 as per live documentation by INTEL.

upon reading various issues that have have been closed. 

I have also copied:
pybackend2.cpython-36m-aarch64-linux-gnu.so.2.32.1
pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.32.1

into `usr/local/lib/python2.7/site-packages`
as well as in the same directory as the python app I intend to run. ' librealsense/wrappers/python/examples"

then run an of the example
`python opencv_pointcloud_viewer.py`

error

```
  File "opencv_pointcloud_viewer.py", line 32, in <module>
    import pyrealsense2 as rs
ImportError: /home/r/librealsense/wrappers/python/examples/pyrealsense2.so: undefined symbol: PyInstanceMethod_Type
```

lramati, 2020-02-20:
Please open a new issue instead of commenting on a years old closed one.
Make sure to indicate what version of python you are using, the various
commands and outputs you've included appear to have both python 2 and 3

>

lauritowal, 2021-01-22:
it workes for me with python3 on a TX-2 after following what @gavan1 did and then adding also `librealsense/build/wrappers/python` to the PythonPath: `PYTHONPATH=$PYTHONPATH:/**<ENTER_YOUR_PATH_TO_LIBRESENSE_HERE>**/librealsense/build/wrappers/python` :)

Gazzamahardhika, 2022-03-30:
hi @MartyG-RealSense @RealSense-Customer-Engineering   im from indonesia ,my english was bad im sorry . i have same problem , i use
rpi4 4gb 
ubuntu 20 /21
realsense t265
python 3.9.7
5.13.0-1017-raspi

i try the pip3 instal pyrealsense2 and its canot work, there a couple refrensse but they use windown operation , can u help me how to install the pyrealsense2 ?  thank u have a nice day

MartyG-RealSense, 2022-03-30:
Hi @Gazzamahardhika  I am not involved in handling T265 related questions so I cannot provide advice specifically about T265.  I will try to provide guidance about installation of pyrealsense2 on Raspberry Pi though.

The **pip install pyrealsense2** packages are designed for x86 / x64 processor chips like those in PC desktops and laptops and are not compatible with the Arm processor architecture in Raspberry Pi boards.

Installing the pyrealsense2 wrapper on Raspberry Pi is not straightforward for either Ubuntu 20.04 or Raspberry Pi OS, unfortunately.  

Would it be possible for you to use Raspbian / Raspberry Pi OS instead of Ubuntu and build librealsense and the Python wrapper using the **Acrobotics** guide for Pi 4 in the link below?

https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

*************

If you need to install on Ubuntu, an alternative approach may be to build librealsense and pyrealsense2 together from source code using an Nvidia Jetson guide at [the link below](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049) (as Jetson boards also use Arm processors).

You would have to remove the CMake build term **-DBUILD_WITH_CUDA:bool=true** from the CMake build instruction, as Pi boards cannot use CUDA graphics acceleration.  You would also have to change **-DPYTHON_EXECUTABLE=/usr/bin/python3.6** to reference python3.9 instead.

***********

A third approach could be to use the **libuvc backend** installation method that has worked consistently with Raspberry Pi in the past.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The problem with this approach is that its build script only builds librealsense but not the pyrealsense2 wrapper.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

Conceivably, you could input the commands in the script manually one by one and when you get to the part where you input the CMake instruction, instead of using this one:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

Use this one:

`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.9 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

Gazzamahardhika, 2022-04-01:
> 

hey @MartyG-RealSense its solved ,thank u so muchi had my pyreal2 noww ,wish u all the best brodi

MartyG-RealSense, 2022-04-01:
That's excellent news, @Gazzamahardhika - thanks very much for the update and the kind words  :)