realsenseai/librealsense #2587: Purpose of make_shared pointer to Pipeline? [C++]

HippoEug, 2018-10-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |             NIL                          | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

# Introduction
Hello, I am trying out the rosbag example using C++, in particular `rs-record-playback`. However, there are some code changes that I fail to understand.. I hope someone can clarify this for me! 😃 

# Issue 1
#### `rs-record-playback.cpp`
```
auto pipe = std::make_shared<rs2::pipeline>();
pipe->start();
```

#### `rs-capture.cpp`
```
rs2::pipeline pipe;
pipe.start();
```

My question is, what is the difference between these 2 ways of writing the Realsense SDK pipeline?
From the [record-playback GitHub README](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback), it says "In order to allow using different types of devices (rs2::recorder, rs2::playback) and read\write to the same file, we define a shared pointer to rs2::pipeline. This way we'll be able to release previous resources when switching to a different device". 

Here is where I am confused, why is `rs2::recorder` or `rs2::playback` a device per se? Aren't they supposed to behave more like functions? Isn't the device the single Realsense D435 camera in my case?

# Issue 2
This is a syntax issue, similar syntax to the shared pointer (Issue 1).
#### `rs-record-playback.cpp`
```
if (!device.as<rs2::playback>())
{
    frames = pipe->wait_for_frames(); // wait for next set of frames from the camera
    depth = color_map(frames.get_depth_frame()); // Find and colorize the depth data
}
```
I am confused by `(!device.as<rs2::playback>())`.. is there any way of explaining this in simpler or layman terms? I have never written in this syntax at all..

It was only recently that I understood that:
```
window(int width, int height, const char* title)
        : _width(width), _height(height)
```
is equivalent to 
```
window(int width, int height, const char* title) {
		_width = width;
		_height = height;
```

Thank you in advance! Good day 😸

dorodnic, 2018-10-24:
Hi @HippoEug 

Both `record` and `playback` classes derive from common base class - `device`. Perhaps less confusing names would be `recorder` and `player`. The idea is for these two "devices" to behave just like a regular device, so you can write your code once, as if it was always running on real hardware. 

If at any point you need to check if you are running on real hardware or a recording, you can check if `device.is<playback>()` is true. If you know that you **are** running from a recording, you can access additional functionality by calling `device.as<playback>()`. You can think of it as `(playback)device` but safe. I'm not 100% sure why in the if statement the author chose to use `as` but I hope the purpose of these functions is more clear now.

Regarding the question with the `shared_ptr` - as long as you have a `playback` or `record` object, the underlying `*.bag` file remains open and locked for writing. 
In the `playback-record` example the device is managed by the `pipeline` and you shouldn't technically rely on when the pipeline is going to release the device. Maybe it will close the old file first and then open the new file, and maybe the other way around. 
That's why the easiest way to get around that is just to replace the `pipeline` object. I don't remember why a simple `pipe = rs2::pipeline()` wasn't sufficient, but the point is, you need to release the pipeline object to make sure the file is no longer locked. 

Hope this helps

RealSense-Customer-Engineering, 2018-10-26:
[Realsense Customer Engineering Team Comment]
Hi HippoEug,

Do you still need help on this?

Thanks!

HippoEug, 2018-10-29:
Hello there, everything is clear. You can close this now, thank you!

closed 2018-10-29 by RealSense-Customer-Engineering