realsenseai/librealsense #2590: is there anything CSP function in the librealsense's internal

zhouwg, 2018-10-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D410, D415 } | 
| Operating System & Version |   Linux (Ubuntu 14/16/17)   |                                | 
| Platform                                 |  PC Linux |
| SDK Version                            |  { 2.16.1}                          | 
| Language                            |  c/c++ }                          | 

### Issue Description
as a dedicated library for Intel's Realsense depth camera,  is there anything CSP(color space conversation) function provide to developers? it might be very helpful for the developers with librealsense? 

such as:

cvMat -> rs2::frame
blitRSFrame(void *dstSurface, rs2::frame...)  
stretchBlitRSFrame(void *dstSurface, RECT  *dstRect, rs2::frame);
...
...

Thanks very much!

dorodnic, 2018-10-24:
Hi @zhouwg 
Can you please explain what you mean by CSP functions? My first guess was that you are referring to YUY-RGB conversions, but I'm not sure given the examples you provided.

zhouwg, 2018-10-25:
Hi @dorodnic ,
Thanks your comments. your guess is correct!

for example in x264 project:

http://git.videolan.org/?p=x264.git;a=blob;f=filters/video/resize.c;h=dda5db99d9e5108430793e961bdb8e5051596e52;hb=refs/heads/master#l53

if librealsense could export some color space conversation function as following:

typedef struct  {
    int32_t width;
    int32_t height;
    int32_t stride;
    int32_t format;
    void* pixels;
} rs_surface;

typedef struct {
     int32_t x; 
     int32_t y;
     int32_t w;
     int32_t h;
} rs_rect;

blitRSFrame(rs_surface *dstSurface, rs2::frame...)
stretchBlitRSFrame(rs_surface *dstSurface, rs_rect *dstRect, rs2::frame);

it will be more convenient for  the developers which using librealsense.

Thanks!

closed 2018-10-26 by zhouwg