realsenseai/librealsense #2593: Unity AlignmentSample Scene Crashes

SeoHyunSeok, 2018-10-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.16.1 }                          | 
| Language                            |  {C#/unity }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hello I'm using 2.16.1 sdk, and 2018.2.12f1 Unity.

when I click to open AlignmentSampleScene in samples, Unity just crashes.
I found the closed issue, it said try 2.14.0 sdk, so I tried, but still Unity Crashes.

I even went down to 2.13.0 but it doesn't have alignment sample.
I really need alignment functions in my project.

Has anyone solved this crashing issue?
Thanks :)

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
Hi @SeoHyunSeok,

Is this file missing in your working copy?
https://github.com/IntelRealSense/librealsense/commits/development/wrappers/csharp/Intel.RealSense/Option.cs

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
Hi @SeoHyunSeok,

The *.cs files in <librealsense>/wrappers/charp/ are not Unity script.

Build librealsense from source, please follow this document.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

And please make sure turn on the option *BUILD_UNITY_BINDINGS* in CMake.

build the whole project. and then the Unity example project is stored in <librealsense>/wrappers/unity/

SeoHyunSeok, 2018-10-25:
> [Realsense Customer Engineering Team Comment]
> Hi @SeoHyunSeok,
> 
> The *.cs files in /wrappers/charp/ are not Unity script.
> 
> Build librealsense from source, please follow this document.
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md
> 
> And please make sure turn on the option _BUILD_UNITY_BINDINGS_ in CMake.
> 
> build the whole project. and then the Unity example project is stored in /wrappers/unity/

Working very good. Thanks!

closed 2018-10-25 by SeoHyunSeok