realsenseai/librealsense #2595: Problem to install Pyrealsensor in python 2.7

RZz96, 2018-10-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want yo change the exposure in depth image and in order to do that i have to install pysensor.
it give following error:

Collecting PySensors

  Using cached https://files.pythonhosted.org/packages/4e/11/c17454160e80a60587adcb511b760a6ddf8b2d60683bb0edd85919199adf/PySensors-0.0.3.tar.gz

    Complete output from command python setup.py egg_info:

    Traceback (most recent call last):

      File "<string>", line 1, in <module>

      File "c:\users\appdata\local\temp\pip-install-fhhqvm\PySensors\setup.py", line 4, in <module>

        import sensors

      File "sensors\__init__.py", line 19, in <module>

        SENSORS_LIB = CDLL(LIB_FILENAME)

      File "c:\users\appdata\local\continuum\anaconda2\lib\ctypes\__init__.py", line 366, in __init__

        self._handle = _dlopen(self._name, mode)

    TypeError: expected string or Unicode object, NoneType found
    ----------------------------------------

Command "python setup.py egg_info" failed with error code 1 in c:\users\appdata\local\temp\pip-install-fhhqvm\PySensors\

dorodnic, 2018-10-24:
I suspect what you need is [pyrealsense2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#installation): `pip install pyrealsense2`
I'm not sure what is PySensors.

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
Hi @RZz96,

Still use "pyrealsense2". Try the code like below. By the way, which camera do you use?

**import** pyrealsense2 **as** rs
pipeline = rs.pipeline()
profile = pipeline.start(config))
depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(rs.option.exposure, <exposure value>)

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
Hi @Rzz96,

If no further support required for this topic, will close it.

closed 2018-10-30 by RZz96