realsenseai/librealsense #2604: Force grab of a new frame via software trigger

jimwormold, 2018-10-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Firmware Version                   | (05.10.03.00) | 
| Operating System & Version |   {Linux 18} || 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.16.1 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
I'm using a D415 to capture some near infrared images synchronised with lighting (at 30 fps) via a software trigger. My problem appears to be that the first image that I request via wait_for_frames() is one that has been taken before I have asked for it (ie before a light has been switched on). I assume that the images are placed on a stack and then returned when a call to wait_for_frames() is made.

The order goes:
1. Turn on LED1, and await ack
2. call wait_for_frames() and add image to list
3. Turn off LED1 and turn on LED2 and await ack
4. Call wait_for_frames() and add image to list

Assuming that my above logic is correct in explaining the reason that the first frame doesn't show the LED lit, is there a way of forcing the D400 to return the next new frame, rather than one it already has on its stack. Or alternatively is it possible to test the age of frame returned to see if it is stale or not, ie if its more than 0.02s old, then grab another one.

Many thanks

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@jimwormold Could you please provide a picture of your test setup? Thanks!

jimwormold, 2018-10-31:
Thanks for getting back to me.

This is the simple test rig that I use to check syncing. 3 LEDs that switch on/off sequentially with each acquisition.

[image]

This is the actual rig on which I wish to use the D415 once the syncing is working satisfactorily with the 8 LEDs in a ring around the camera

[image]

So we get:

[image]
[image]
[image]

Where the first image, should have the LED that is lit up in the second image, and the second image should show what the third image is showing. Timings of retrieving each frame are:

4.1484832763671875e-05 s
0.03150606155395508 s
0.027814388275146484 s

Where we can see that the first one returns instantly, and the others return in a more expected timeframe for 30fps.

Let me know if you need anything else.

RealSense-Customer-Engineering, 2018-11-02:
[Realsense Customer Engineering Team Comment]
Hi @jimwormold,

The function wait_for_frames is not the grab function. It just read the new frame from the queue.

In this case, it seems you want to sync between the camera and LEDs, frame by frame.

I think you can read the "External Trigger" section in Multi-camera Configurations White Paper on our website (https://realsense.intel.com/).

https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdf

RealSense-Customer-Engineering, 2018-11-20:
[Realsense Customer Engineering Team Comment]
@jimwormold, does this document solve this issue?

closed 2018-11-23 by RealSense-Customer-Engineering