realsenseai/librealsense #2605: I get an error when I read a depth image on the RK3399 platform

McJun1, 2018-10-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |  Ubuntu16.04  | 
| Kernel Version (Linux Only)    |  4.4.138 | 
| Platform                                 | RK3399 |
| SDK Version                            |  2.16.1  | 
| Language                            |  {C/C++ }                          | 
| Segment			|  Robot  | 

I get an error when I read a depth image of 1280*720/640*480 resolution on the RK3399 platform.I set the frame rate to 30hz, but sometimes it only came out 1 frame per second, and even reported a "Frame didn't arrived within 5000" error, obviously he was stuck. I checked the cpu occupancy rate, less than 10%. I checked my USB3.0 transfer rate, which reached 187M/s. Obviously this is not a USB transfer problem. Then I checked my kernel error and found the following LOG:
[37104.965447] Call trace:
[37104.965671] Exception stack(0xffffffc0d5733a00 to 0xffffffc0d5733b30)
[37104.966233] 3a00: ffffffc0d5733d10 0000008000000000 ffffffc0d5733bd0 ffffff8008867f58
[37104.966921] 3a20: ffffff8009257078 0000000000000002 ffffffc0d5733a50 000000020003ec6c
[37104.967609] 3a40: 0000000000000000 0000000100000000 ffffffc0d5733af0 ffffff80080ee30c
[37104.968297] 3a60: ffffffc0d5733b50 ffffff8008eac1ea ffffff8008ca6d30 ffffffc0d9834018
[37104.968985] 3a80: 0000000000000000 ffffffc0d9cb2540 ffffffc0d5733d10 ffffffc0e30f3900
[37104.969673] 3aa0: 000000000000001e 00000040eeed2000 ffffffc0f7f58320 0000000000000000
[37104.970361] 3ac0: 0000000000000000 0000000000000000 ffffff8009256845 65786970206e776f
[37104.971050] 3ae0: 2074616d726f666c 0000000005f5e0ff ffffff800925681f 00000000fffffffe
[37104.971738] 3b00: 0000000000000030 000000000000000a 00000000000cee4e 0000000000000000
[37104.972426] 3b20: ffffff80081b8280 0000007f7d738ea0
[37104.972860] [<ffffff8008867f58>] v4l_enum_fmt+0xfbc/0x1424
[37104.973341] [<ffffff800886a210>] __video_do_ioctl+0x1f0/0x26c
[37104.973844] [<ffffff8008869f28>] video_usercopy+0x450/0x4fc
[37104.974335] [<ffffff800886a010>] video_ioctl2+0x3c/0x4c
[37104.974792] [<ffffff8008863a5c>] v4l2_ioctl+0xa8/0xb8
[37104.975239] [<ffffff80081b7dd4>] do_vfs_ioctl+0x9c/0x548
[37104.975707] [<ffffff80081b82d0>] SyS_ioctl+0x50/0x80
[37104.976142] [<ffffff8008082f30>] el0_svc_naked+0x24/0x28
[37104.981694] uvcvideo: Non-zero status (-71) in video completion handler.

RealSense-Customer-Engineering, 2018-10-29:
[Realsense Customer Engineering Team Comment]
@McJun1 Please try with USB2.0 to see if you still have issue. For USB3.0, need apply UVC patch for UVC optimization to get the stable streaming. You can check with rockchip to get more details about the UVC patch. Thanks!

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
@McJun1 Did you get chance to try USB2? Any other questions for this ticket? Looking forward to your reply. Thanks!

RealSense-Customer-Engineering, 2018-11-12:
[Realsense Customer Engineering Team Comment]
@McJun1 Any update for this ticket? If no other questions, we'll close the ticket in a few days. Thanks!

RealSense-Customer-Engineering, 2018-11-16:
[Realsense Customer Engineering Team Comment]
@McJun1 Didn't get update from you. Will close this ticket at this moment. If you still have issue, please re-open or create a new ticket. Thanks!

closed 2018-11-16 by RealSense-Customer-Engineering

McJun1, 2018-11-16:
I try the usb2.0,but it can't work

RealSense-Customer-Engineering, 2018-11-16:
[Realsense Customer Engineering Team Comment]
@McJun1 Please try PR https://github.com/IntelRealSense/librealsense/pull/2489 to build Android sample and try again.

RealSense-Customer-Engineering, 2018-11-27:
[Realsense Customer Engineering Team Comment]
@McJun1 Did you get through now? Please update. Thanks!