realsenseai/librealsense #2607: error by storing frames in a loop into a list

CasimirBo, 2018-10-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win(10)   | 
| Platform                                 | PC  |
| SDK Version                            |   2.16.1   | 
| Language                            |  python  | 
| Segment			|  others  | 

### Issue Description
<If I poll for Frames in a Loop and store the depthframe at each itteration into a list, the first 18 objects look good. After this every next frame seems to be the same as the 16th Frame. So the framenumber, what I print ist the same. 

Here is my code for this:
[image]
And the output:
[image]

If I store the depthframe to "a" and not to the list, everything is okay. 

And if I try to store the framest into a list, the first 16 frameset works and after that the program is just working, but nothing ist happening.

Thanks for your support.

Kind regards,
Casimir
 >

dorodnic, 2018-10-25:
Hi,
Can you try adding `frameset.keep()` in the loop?
Unless you explicitly request it, we keep finite pool of 16 frames of each type.

CasimirBo, 2018-10-25:
Hi,
that simply solved the problem.

Thanks a lot

closed 2018-10-26 by RealSense-Customer-Engineering