realsenseai/librealsense #2611: No framed received in ubuntu 16.04, 18.04

ljc19800331, 2018-10-26:
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer) | 
| Operating System & Version |   {Linux (Ubuntu 14/18)  | 
| Kernel Version (Linux Only)    |  (4.15.0-36-generic)  | 
| Platform                                 | PC |
| SDK Version                            |  {librealsenseSDK2 }                          | 
| Language                            |  {C++/Python}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
All the setups for the realsense (including realsense-viewer and other good applications) are good before I reinstalled the system and update for ubuntu 18.04. I faced the problem that I could not received frame with realsense-viewer ("no frame received"). 

I tried with the usb 3.0 ports with both ubuntu 18.04 (in my laptop) and ubuntu 16.04 in my PC. However, I faced the following problems:
1. in ubuntu 18.04 laptop, the realsense-viewer cannot received frame.
2. in ubuntu 16.04 PC, I could not Cmake the librealsense library with the following errors:

Download error
http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.3452/target-0.0.18.3452.mvcmd
-- Configuring incomplete, errors occurred!
See also "/home/maguangshen/librealsense/build/CMakeFiles/CMakeOutput.log".

3. I checked with the usb 3.0 port, it is working good in windows 10 (I installed dual system).
4. I checked with the usb and uvc in linux: 
  
Bus 002 Device 006: ID 0951:1666 Kingston Technology DataTraveler G4
Bus 002 Device 004: ID 0461:4d81 Primax Electronics, Ltd Dell N889 Optical Mouse
Bus 002 Device 003: ID 04b3:3025 IBM Corp. NetVista Full Width Keyboard
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 007: ID 8086:0ad3 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 058f:6362 Alcor Micro Corp. Flash Card Reader/Writer
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

uvcvideo               90112  0
videobuf2_vmalloc      16384  1 uvcvideo
videobuf2_v4l2         24576  1 uvcvideo
videobuf2_core         40960  2 uvcvideo,videobuf2_v4l2
videodev              180224  3 uvcvideo,videobuf2_core,videobuf2_v4l2
media                  40960  2 uvcvideo,videodev

I found the relevant issues but could not find any good solutions. Thanks.

RealSense-Customer-Engineering, 2018-10-26:
[Realsense Customer Engineering Team Comment]
Hi @ljc19800331,

1. In Ubuntu 18.04, can your D415 be recognized as a USB3 device? Please run "usb-devices" to show the USB device info. Do you use the pre-built realsense-viewer or build it from the source?
2. What's D415 firmware version?
3. Please check if you enable "BUILD_WITH_TM2" in CMAKE by accident?

dorodnic, 2018-10-26:
A quick comment regarding @RealSense-Customer-Engineering point no. 3 - it is not your mistake, BUILD_WITH_TM2 is true, you can work-around the problem by setting it to false (`cmake ../ -DBUILD_WITH_TM2=false`) You are most likely behind proxy, please consider checking your network settings, since we do plan to continue distributing binaries in this way via CMake.

ljc19800331, 2018-10-26:
Hello @RealSense-Customer-Engineering , 
1. I am using the prebuilt realsense-viewer after installing the packages sudo apt-get install librealsense2-dkms, sudo apt-get install librealsense2-utils. 
2. Yes sure, this is the usb-devices information when I connect the D415:

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 3
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.15
S:  Manufacturer=Linux 4.15.0-36-generic ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=0000:00:1a.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480 MxCh= 6
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=0024 Rev=00.00
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=01 Dev#=  3 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=058f ProdID=6362 Rev=01.00
S:  Manufacturer=Generic
S:  Product=Mass Storage Device
S:  SerialNumber=058F63626420
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 3
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.15
S:  Manufacturer=Linux 4.15.0-36-generic ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=0000:00:1d.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480 MxCh= 8
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=0024 Rev=00.00
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=02 Prnt=02 Port=01 Cnt=01 Dev#=  3 Spd=1.5 MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=04b3 ProdID=3025 Rev=01.09
S:  Manufacturer=LITE-ON Technology
S:  Product=USB NetVista Full Width Keyboard.
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=70mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid

T:  Bus=02 Lev=02 Prnt=02 Port=02 Cnt=02 Dev#=  4 Spd=1.5 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0461 ProdID=4d81 Rev=02.00
S:  Product=USB Optical Mouse
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=98mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.15
S:  Manufacturer=Linux 4.15.0-36-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:04:00.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 4
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev=04.15
S:  Manufacturer=Linux 4.15.0-36-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:04:00.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=04 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  3 Spd=5000 MxCh= 0
D:  Ver= 3.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=8086 ProdID=0ad3 Rev=50.92
S:  Manufacturer=Intel(R) RealSense(TM) 415
S:  Product=Intel(R) RealSense(TM) 415
S:  SerialNumber=821413024309
C:  #Ifs= 5 Cfg#= 1 Atr=c0 MxPwr=440mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 3 Alt= 0 #EPs= 0 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 4 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo

3. The D415 frame version is 05.09.02.00, I would like to update but some problems happen.
4. Yes I could build the files after following @dorodnic 's instruction on cmake ../ -DBUILD_WITH_TM2=false but still could not get the frame working ("Frame didn't arrived within 5000").

dorodnic, 2018-10-26:
Well the proxy settings really depend on your network situation. You should be able to reach [target-0.0.18.3452.mvcmd](http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.3452/target-0.0.18.3452.mvcmd) from browser and via CMake. But no worries, disabling `DBUILD_WITH_TM2` has no negative side-effects for  D400 users, and we have plenty of time to streamline T265 deployment process.

dorodnic, 2018-10-28:
@RealSense-Customer-Engineering - can we check if firmware 5.9.2 was signaling the "end of frame" bit?
(I know 5.8.15 was not, and 5.9.2 was bundled with LRS 2.9.0 so its quite old)
In latest DKMS we tightened the rules for UVC compliance (details here - #2341) and it will result in "no frames received" behavior on old firmware. 

@ljc19800331 @mwrightE38 please consider updating your camera firmware by following [Linux-RealSense-D400-DFU-Guide.pdf](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf) guide. 

I know its a hassle to install, but there are actually objective reasons to upgrade. We are constantly dancing around USB3 and USB2 bandwidth limitations, and the new firmware improves the stability of streaming depth and color at high resolutions. Alternatively, you are safe to go back to previous version of DKMS and it also should solve the problem.

ljc19800331, 2018-10-28:
@dorodnic Thanks for the information. I think the problem is because of the latest update on UVC. I tried with the method in Linux-RealSense-D400-DFU-Guide.pdf guide and found the following problem: 
age_UVC_5_10_3_0.bin 
D400 busnum = 2
D400 devnum = 2
D400 FW file path = /home/maguangshen/Downloads/Signed_Image_UVC_5_10_3_0.bin
Could not find device in opmode: Device found, but incorrect (vid, pid) combo. Found (8087, 24)
Device pid = 36
Valid pid not found

I tried with the similar method presented in #1986 and I have checked the connection is a usb3.0 port. 

Thanks for the answer.