realsenseai/librealsense #2612: [ USB SCP/CAM overflow ],  [ RT IC2 Config error ]  with D435

MemoryNode, 2018-10-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 5.10.06.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.16.1                          | 
| Language                            |  C++ / VS2015                           | 

### Issue Description
Hi~

We have been trying the long-term running test with D435 in 1280*720 resolution and 30fps(using the black cable that came with the camera and plug it into the USB 3.0 of computer directly ), but unfortunately it would be crashed in a few minutes;
[image]
After running the Realsense viewer app, we have discovered that the same instabilities exist there, and some of the flowing exception message would be throw out: 
| USB CAM overflow
| USB SCP overflow
| RT IC2 Config error 
| Flag B-laser disabled 
| Right MIPI error
| "Error polling loop is not behaving as expected!  This can indicate an issue with camera firmware or the underlying OS..."
[image]
[image]
[image]
[image]
[image]

Every crash could cause the sensor to be disconnected;

In case of abnormal crash, and with the cable was not unplugged, there are several scenarios:
1. Both the Depth Quality Tool and Viewer app were still connected with the D435, and the IR Projector was still running, but No Frames Received;
[image]
[image]

2. Rerun the Viewer or Depth Quality Tool so it can work properly again, but the same thing happens a few minutes or more later, leading to a crash;

3. The Depth Quality Tool can still receive frames from D435, but the data displayed is extremely messy, and the distance calculated is even negative；
[image]
[image]

4. Neither the Viewer nor the Depth Quality Tool can detect the existence of D435, and at this point, the IR projector of D435 was still working (you can see a red point);

We now have four D435 sensors, and issues will happen on my computer, but on other computers, it seems eveything is ok, no exception, no crash; ( Maybe I was lucky enough to see these issues =_=!!! )

I would appreciate any clues or Suggestions.

HippoEug, 2018-10-26:
Does your computer happen to be older? I tried using the D435 on a pretty old laptop running Windows 10, and experienced `USB CAM overflow` and `USB SCP overflow` frequently. Suspected it was due to the crappy USB, and ended up switching to another laptop. I do not have this issue anymore..

Not sure what is causing it exactly though :/

MemoryNode, 2018-10-26:
@HippoEug 
I thought about that possibility, But I didn't have this problem when I tested it on other computers of the same model...
( Dell OptiPlex 7040MT , Intel CORE i5 )

MemoryNode, 2018-10-26:
@dorodnic 
Thanks for your reply.

My computer OS is [Win10 Home Chinese version], and the D435 is directly connectes to one of the USB3.0 Ports on the motherboard. 

Interestingly, sometimes it can also run for a long time(Up to 5 hours, and I didn't do anything), but usually only 10-30 minutes.

Just now, it crashed again( USB SCP overflow | RT IC2 Config error ), the Viewer were still connected with the D435, i try trun on the Stereo Module, it works fine, i can see the depth image, but when i continue trun on the RGB Camera, the Viewer app jammed, and then crashed ...
[image]

HippoEug, 2018-10-26:
> Interestingly, sometimes it can also run for a long time(Up to 5 hours...)

Do you mean turning on and leaving the pipeline on (streaming) continuously for 5 hours straight? My D435 feels hot to the touch if I leave it on for more than 20 minutes..

MemoryNode, 2018-10-26:
@HippoEug 
The same to mine,  It's perfect for hand warmers, especially now that it's getting colder.
With the Depth Quality Tool , we can see that the Asic and Projector Temperature are high. In later products, we will consider adding cooling system to reduce its temperature

MemoryNode, 2018-10-31:
Hi~ @RealSense-Customer-Engineering 
Thank you for your reply.

I did tried to stream both Depth and RGB in a lower resolution and fps(640*480_30fps) for a short time(about 40min), it seems ok. No issues. But we need D435 working at 1280*720, So i am trying to run it for a longer time in 1280*720 and 15fps. I will share any progress with you in time.( Maybe this bad thing shouldn't be called sharing... )

The computer I mentioned above was the same model as my computer, but the version of windows10 was different, My operating system is windows10 Home Chinese version, but they do have other versions. Could this be the cause of the issues?

I'll do more types of comparison tests, but reinstall Windows now seems impossible. Now that you mention the driver, is there some of the driver-related exceptions that are already known? If you know , please let me know and I will be very very grateful.

MemoryNode, 2018-11-01:
@RealSense-Customer-Engineering 

I tested two resolutions(1280 * 720 _ 15 fps  /  640 * 480 _ 30 fps), and at either resolution, these issues happened again...

And, in my own application, there are exceptions like this:
[image]
[image]

Is this also due to the [ USB SCP overflow ] error ?

RealSense-Customer-Engineering, 2018-11-05:
[Realsense Customer Engineering Team Comment]
Hi MemoryNode,

We understand that it is not a simple work to refresh a Window installation, and the only Windows driver we would suggest to try is: https://downloadcenter.intel.com/download/27799/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10.
But there is not much we can help on the problem happening on your computer given other same computer models working well.

This ticket will be closed soon if there is nothing else.

Thank you!

MemoryNode, 2018-11-06:
@RealSense-Customer-Engineering 

Thank for your reply.

I got a new Type-C cable, and tried with it yesterday, it seems ok (about 3 hours),  Maybe it's just an accident, and I need more tests. 

Thank for your help again, @dorodnic @RealSense-Customer-Engineering , if there are other questions , i will open a new issue for it.

closed 2018-11-06 by MemoryNode

jungladicitta, 2019-08-08:
> I got a new Type-C cable, and tried with it yesterday, it seems ok (about 3 hours), Maybe it's just an accident, and I need more tests.

how was your tests? did new cable helped? everything works fine?

CaleMcCollough, 2019-09-01:
Sadly updating the webcam driver only reduces the frequency of but does not eliminate the freezing. I've read on the internet that Windows 10 has a problem with their change to YUY2 color space breaking webcams. The Windows 10 Camera is completely unusable; I can't even load the app right now, but in the RealSense viewer I can change to YUYV color space and I'm not getting the USB CAM overflow. Maybe a fresh install of Windows 10 is required to fix some artifact default settings from the previous update. The D435i has frozen in OBS since the discovery of this fix; if I find a fix I'll post it here.

CaleMcCollough, 2019-09-22:
I forgot I have an Intel Laptop and I just installed my D435i on my Intel Laptop and it works just fine in OBS without the drivers so that's not the cause of the USB Cam Overflow error. I've concluded that it's an undercurrent Issue because the product won't work when connecting to a hub with 2 USB 3 connections for power and data, but it will when plugged into an external power supply. The only good explanation that it's an undercurrent problem related to #2045 that is just causing the USB hub to go down and not the mobo USB chipset. I have another 3 foot USB 3 C cable that won't work, so it's not the cable length, and the camera doesn't hog up the entire USB 3 bus and power is the typical culprit. I'm going to order a cable meant for charging a phone and a PCI-X USB 3 card and report back.

SamGreen9, 2020-01-22:
@cale-mccollough, I am experiencing similar issues to the one you describe. Can you tell me if you were ever able to resolve your issues? Did using a PCI-X USB chipset improve things?