realsenseai/librealsense #2617: Android Camera Permission

indigohedgehog, 2018-10-26:
What is the specific issue that causes to need a rooted device?

dorodnic, 2018-10-26:
This part:
```shell
su
setenforce 0
lsusb
chmod 0777 /dev/bus/usb/<Bus number>/<Dev Number>
```
from [AndroidJavaApp.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidJavaApp.md).

There are several alternatives, including building a custom image as proposed in some pull-requests (#2489). We are still evaluating this method, but from my personal experience, building Android image from source is not something I as a developer would do at home. 

Some people work-around this issue by requesting permissions at Java level and passing the resulting file-descriptor to libusb (see [UVCCamera](https://github.com/saki4510t/UVCCamera)). We evaluated this method and chose not to endorse it, for two reasons - 
1. There is no guaranty Google wont shut-down this method at some point. 
2. This is not a built-in capability of libusb, thus requires maintaining a custom version of libusb.

Since librealsense is permissively licensed and open-source, you are free to pursue this approach, including commercially, but we can't offer support. 

We are also looking into other methods. 

Intel also has a dedicated implementation of the camera HAL it shares with its partners, but this is also not really suitable for dev-kits.

zhouwg, 2018-10-26:
I'm also interesting with this question. appreciate too much @dorodnic 's detailed elaboration.
is there any planning or roadmap of librealsense's supportive with Android ? 
or what's the planning or roadmap of librealsense's supportive with Android? 

we both know that Intel's Realsense Camera is an imagination and creative product and the librealsense is also a excellent c++ library. and we also know that there are huge Android based devices in the mobile market or IoT market, it will be very helpful if we could get more clear planning/roadmap of librealsense.

indigohedgehog, 2018-10-26:
I think the approach for handling communication for android, is with UsbManager class, instead of libusb, besides it should stop using libuvc at all.

To properly support android, this means replacing libuvc, and by doing that, you can't use stream class directly, therefor all the content of libuvc folder has to be reimplemented in android. Then in the native lib, orchestrate an input/out flow.

closed 2018-10-26 by indigohedgehog

zhouwg, 2018-11-03:
Thanks the detailed expatiate.  I have following questions:
1) could we maintain a custom libusb for librealsense?  then invoke the standard Android Java API in native layer to obtain the required infos for librealsense. 

the following is a simple example to illustrate the possibility of the above proposal:
the apk would preview bag file on non-rooted Android phone:

 [https://github.com/zhouwg/irsa/blob/master/img-folder/irsa_example_on_non_rooted_phone.jpg](https://github.com/zhouwg/irsa/blob/master/img-folder/irsa_example_on_non_rooted_phone.jpg)

2) it seems there are two major issues with librealsense on Android device:  patch Android kernel; root privilege to access /dev/video* and /dev/bus/usb/*.   
   librs could running well on any non-rooted Android device  if Google could provide a customize  AOSP which add the librealsense kernel patch in Android kernel and add some SELinux  "door" for librealsense.