realsenseai/librealsense #2618: Updating to firmware 5.10.6 prevents connecting to two cameras

carun, 2018-10-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |   {Win 10)| 
| Platform                                 | PC  |
| SDK Version                            |  2.16.2 }                          | 
| Language                            |  {C/opencv}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I understand that this is a project specifically focusing on librealsense SDK. However I don't see any other place where I can report this.

We recently updated our production systems with [production version of firmware 5.10.6](https://downloadcenter.intel.com/download/28237/Latest-Firmware-for-Intel-RealSense-D400-Product-Family).

After the update the system fails to detect the cameras very often and it renders the system unusable.

The cameras are not detectable even by the firmware update tool.

It is unfortunate to see such issues with production grade firmware. Please let us know how to resolve this.

carun, 2018-10-26:
Posted on the [community forum ](https://communities.intel.com/message/574311#574311)as well.

carun, 2018-10-26:
FYI, we have downgraded the production systems from firmware version 5.10.6 to 5.9.13. Things are back to normal.

Also SDK 2.16.2 has some bug. The app just crashes in production. I just saw that it's a pre-release. But unfortunately we can't get the stracktrace.

So the working combination is firmware 5.9.13 and SDK 2.16.1.

dorodnic, 2018-10-26:
Hi @carun 
We did identify a problem with the new software release, it should be limited to Windows and mostly just the Depth Quality Tool application. We are working to publish a hot-fix ASAP. Is this the app that crashed in your environment?

carun, 2018-10-26:
@dorodnic It's a Windows app, consuming RGB frames. We don't use depth information.

```c++
        auto frameset = _pipe.wait_for_frames(1000); // Wait for next set of frames from the camera
        rs2::frame rs_frame = frameset.get_color_frame();
        // Query frame size (width and height)
        const int w = rs_frame.as<rs2::video_frame>().get_width();
        const int h = rs_frame.as<rs2::video_frame>().get_height();

        // Create OpenCV matrix of size (w,h) from the colorized depth data
        cv::Mat image(cv::Size(w, h), CV_8UC3, (void*) rs_frame.get_data(), cv::Mat::AUTO_STEP);
        frame = image;
```

closed 2018-12-12 by RealSense-Customer-Engineering