realsenseai/librealsense #2619: ROS bag and timestamps

pedropgusmao, 2018-10-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D434  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-38-generic                                       | 
| Platform                                 | PC   |
| SDK Version                            |  2.16.1                         | 
| Language                            |  C++  | 
| Segment			|  Robot   | 

### Issue Description
Hello,

I am using ROS Melodic Morenia and I am having issues when trying to play the rosbags generated by Intel RealSense Viewer app. The following is what I get if I run ```rosbag play [ROSBAG_FILE]```.
```
[FATAL] [1540565279.391861343]: Character [ ] at element [32] is not valid in Graph Resource Name [/device_0/sensor_0/option/Enable Auto Exposure/value].  Valid characters are a-z, A-Z, 0-9, / and _.
```
Also the start date of the rosbag seems to be off by a bit. This is what I get when I run ```rosbag info``` :
```
start:        Jan 01 1970 01:00:00.00 (0.00)
end:          Jan 01 1970 01:00:02.40 (2.40)
```

dorodnic, 2018-10-26:
Hi @pedropgusmao 
This a known issue that we do want to address (#1161), **carefully** without breaking compatibility for TBs of content. You can overcome it right now by playing the recording via RealSense ROS wrapper - [rs_from_file.launch](https://github.com/intel-ros/realsense/blob/development/realsense2_camera/launch/rs_from_file.launch)
This is probably also the recommended way since it will ensure you get the same topics structure as from real device, regardless of internal format.

pedropgusmao, 2018-11-06:
Thanks!

What if I launch the **rs_aligned_depth.launch** (from https://github.com/intel-ros/realsense) and record topics using regular **rosbag record** ? Will I be able to replay it afterwards with the correct timestamps? Is there a penalty in terms of performance (loss of frames) if I use regular ROS instead of using the Recording from realsense-viewer?   

Sorry if this question actually belongs to the other repository, but I believe you guys to be the closest to the hardware and I need a way to save 60fps for both depth and color at 848x480 with system timestamps.

pedropgusmao, 2018-11-08:
Hello @dorodnic , do you think it is possible to do the above?

closed 2019-01-15 by RealSense-Customer-Engineering

kvigulis, 2019-03-28:
Hi,

I have the same issue with the spaces in the bag file topics. But the thing is that I have some pre-recorded files that I would like to process, so re-recording in not an option. I've tried using `rs-covert` to extract the images files, but for both of my bag files it crashes midway. What do you think would be the best way to extract the depth and color images, using some other means?

I'm only interested in extracting data from `/device_0/sensor_0/Depth_0/image/data ` and `/device_0/sensor_1/Color_0/image/data `.

Thank you

wafeez, 2020-04-22:
HI, 

As the given bag data has spaces in the bag, do you have other sample bag data for me to test SLAM on the D435i sensor? mainly on pointcloud data.

Your cooperation are appreciated.