realsenseai/librealsense #2620: Can't install realsense viewer on windows 10

umersadiq, 2018-10-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> [Can't install realsense viewer]) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am unable to install realsense-viewer on a windows10 host. My machine is the new intel NUC Hades Canyon. When I download and click on the realsense-viewer.exe file that I downloaded off of github it does nothing. I am not an avid MS windows user and not sure whats going on. Can't find anything on google - any ideas?

MartyG-RealSense, 2018-10-27:
Could you please download the 266 mb file that contains the whole of the 2.16.2 version of the SDK and report back whether the Viewer then works for you, please?  The filename on the downloads page is 'Intel.RealSense.SDK-2.16.2.304.exe'

https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.2

umersadiq, 2018-10-27:
Thanks for getting back quickly…

I installed the SDK.
This installs in the Program Files (x86) folder.

The behavior is the same when i go into tools > realsense-viewer.

MartyG-RealSense, 2018-10-27:
There was a similar case to yours in the past week on the Intel Support website, where nothing happened when that user tried to launch the Viewer.  They fixed it by updating their graphics driver.

https://communities.intel.com/message/573265#573265

umersadiq, 2018-10-27:
That fixed it - Thanks!

closed 2018-10-27 by umersadiq