realsenseai/librealsense #2621: Not able to install librealsense on Ubuntu 16 (4.15.0-38-generic)

Kottapalli-Naveen, 2018-10-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { NA} | 
| Firmware Version                   | (NA) | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0-38)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Facing the below compilation error while installing librealsense on Ubuntu 16 (4.15.0-38-generic).  This error occurred while executing step (./scripts/patch-realsense-ubuntu-xenial-joule.sh) mentioned in the installation guide.

-------------------------------------------------------------------------------------------------------------------------
WARNING: "cpu_tss" [/home/sriram/Downloads/librealsense-master/xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "set_normalized_timespec" [/home/sriram/Downloads/librealsense-master/xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_cleanup" [/home/sriram/Downloads/librealsense-master/xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "copy_in_user" [/home/sriram/Downloads/librealsense-master/xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_init" [/home/sriram/Downloads/librealsense-master/xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_create_link" [/home/sriram/Downloads/librealsense-master/xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
/home/sriram/Downloads/librealsense-master/xenial/drivers/media/usb/uvc//Module.symvers: Permission denied
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[1]: *** [__modpost] Error 1
Makefile:1426: recipe for target 'modules' failed
make: *** [modules] Error 2
make: Leaving directory '/home/sriram/Downloads/librealsense-master/xenial'

dorodnic, 2018-10-27:
Hi @Kottapalli-Naveen 
Just to make sure - are you running on an Intel-Joule board? What installation document are you following?

Kottapalli-Naveen, 2018-10-28:
Hello @dorodnic 

How do I get the board type?  Am using the below installation instructions.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Thanks

dorodnic, 2018-10-28:
The document says:
* Intel® Joule™ with Ubuntu Based on the custom kernel provided by Canonical Ltd.
`./scripts/patch-realsense-ubuntu-xenial-joule.sh`

If you do have a Joule, it would look something like this:
[image]

These are not very common, so I think you probably should be running `./scripts/patch-realsense-ubuntu-lts.sh` instead.

Kottapalli-Naveen, 2018-10-29:
Hello @dorodnic 

Can you please tell how do I get the board type?  Am using the below installation instructions.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Thanks

Kottapalli-Naveen, 2018-10-30:
@dorodnic Things went well with {{./scripts/patch-realsense-ubuntu-lts.sh}}.  Thank you.

closed 2018-10-30 by Kottapalli-Naveen