realsenseai/librealsense #2622: Compile with Visual Studio Express Edition

NeuroRanger, 2018-10-27:
Should the RealSense-Lib be able to compile with VS Express?
I get a missing Header-File called "atlcomcli.h". As far as I know, this is part of the "Active Template Library (ATL)", which is not included in the Express-Version of VS.
I tried installing the WinDDK, etc., adding Include- and lib-Paths like "C:\WinDDK\7600.16385.1\inc\atl71" but even with VS Community-Edition I get missing libraries.

----------------------------------------------------------------------------------------------------

|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | () | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  current master | 
| Language                            |  C/C#/unity | 
| Segment			|  others    | 

### Issue Description
How to compile librealsense SDK with Visual Studio Express Edition

dorodnic, 2018-10-27:
Hi @NeuroRanger 

> Should the RealSense-Lib be able to compile with VS Express?

Yes, it should. At least in theory. The CI we have runs VS Professional, so it is possible we accidentally broke it. For a work-around, if all you need is to develop on top of the SDK, you can use the pre-compiled version ([Intel.RealSense.SDK-2.16.2.304.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.2/Intel.RealSense.SDK-2.16.2.304.exe))

It is possible that the issue can be resolved by installing WDK or some other SDK, but I will mark this as a bug so dev or the support team know to follow-up and double-check. 

Please let us know if you found a solution / work-around.

NeuroRanger, 2018-10-27:
Thanks, dorodnic!

I can propably use the .dll from the SDK-binaries.
But I miss the Unity-Wrapper-Environment, which I already used after compiling the SDK with VS Professional, that worked well since the latest Version-Update. Is there a precompiled Bundle anywhere including the "wrappers/unity"-folder?

Would of course be great to be able to compile it with VS Express.

So thanks for mentioning that for a check.

dorodnic, 2018-10-28:
We publish a [unitypackage](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.2/Intel.RealSense.unitypackage) and the C# wrapper is included in the installer. Hope this can help for now...

smirkingman, 2018-12-07:
Windows 10 + VS 2017 Enterprise, to compile realsense2 you need to add ATL support to VS, as described [https://stackoverflow.com/a/46304843/338101](url)