realsenseai/librealsense #2623: Syncer class for Python Wrapper

General-Ilya, 2018-10-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / **D400** } | 
| Firmware Version                   | (05.10.06.00) | 
| Operating System & Version |   {Ubuntu 16.04} | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.16.2 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {multicam}                          | 

### Issue Description
Will there ever be a syncer class in the python wrapper?

I am working on a multicam setup for gesture recognition, and need to be able to save sync'd footage from the two cameras for training, however it doesn't look like HW sync is possible with the python wrapper.

dorodnic, 2018-10-28:
Hi @General-Ilya 
Sorry, it might be a bit confusing - you can enable hardware sync using `set_option` API. The syncher class provides software sync, bundling everything up at API level. And for now, it can't really help with multi-camera sync (we are working in this direction however). The main problem is that once cameras are in sync, their hardware-timestamps are no longer reliable and the frame-number can be flaky at times (it is not a hardware mechanism but rather firmware counter). So our best recommendation for multicamera for now is to create several pipelines, poll them using `poll_for_frames` and take the latest frameset from each. On Linux this works pretty well, because we have complete control over all queues involved. That said, regularly synchronizing the two cameras clocks is what we plan to do.

General-Ilya, 2018-10-29:
Ah I see; is there any example code I can look at for multicam hardware sync on python? I can't seem to find much of anything on it.

dorodnic, 2018-10-29:
Setting camera to Master - 
```python
Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> pipe = rs.pipeline()
>>> prof = pipe.start()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Camera already streaming
>>> prof = pipe.start()
>>> dev = prof.get_device()
>>> ds = dev.query_sensors()[0]
>>> ds.set_option(rs.option.inter_cam_sync_mode, 1)
>>> ds.get_option(rs.option.inter_cam_sync_mode)
1.0
>>>
```

dorodnic, 2018-10-29:
For Slave - set the same option to 2

General-Ilya, 2018-10-29:
Спасибо!

closed 2018-10-29 by General-Ilya

koonyook, 2019-05-21:
> Setting camera to Master -
>
> ```python
> Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32
> Type "help", "copyright", "credits" or "license" for more [...]

I've found that changing the sync_mode option after starting the pipeline doesn't change the camera behavior until you start the pipeline again.

For me, I do
>>>ctx=rs.context()
>>>devices=ctx.query_devices()
>>>dev=devices[cameraIndex]
>>>device.first_depth_sensor().set_option(rs.option.inter_cam_sync_mode, targetSyncMode)
>>># follow by codes to start your pipeline

xamm, 2022-08-10:
> Hi @General-Ilya
>
> Sorry, it might be a bit confusing - you can enable hardware sync using `set_option` API. The syncher class provides software sync, bundling everything up at API level. And for [...]

Is using poll_for_frames for each device and taking the Last arrived Frame still the recommended approach when using multiple devices and streaming them in python?