realsenseai/librealsense #2629: howCalibration resolution

zhouq123, 2018-10-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

zhouq123, 2018-10-28:
hi, i have a question, my camera model is D415. after using the Dynamic Calibrator software, i can get a file "RgbCalibrationTable.xml", and the parameters about rgb:
    <param name = "ResolutionRGB">
        <value>1920</value>
        <value>1080</value>
    </param>
    <param name = "FocalLengthRGB">
        <value>1388.26</value>
        <value>1388.26</value>
    </param>
    <param name = "PrincipalPointRGB">
        <value>936.129</value>
        <value>544.581</value>
    </param>

but how can i calibration rgb image in different resolution, for example: 640*480.

RealSense-Customer-Engineering, 2018-10-31:
[Realsense Customer Engineering Team Comment]
Hi @zhouq123,

Dynamic calibrator is used to optimize the extrinsic parameters (translation and rotation).
You can run "$ rs-enumerate-devices -c" to show more intrinsic parameters based on different RGB resolution.

zhouq123, 2018-10-31:
> [Realsense Customer Engineering Team Comment]
> Hi @zhouq123,
> 
> Dynamic calibrator is used to optimize the extrinsic parameters (translation and rotation).
> You can run "$ rs-enumerate-devices -c" to show more intrinsic parameters based on different RGB resolution.

Got it, thanks.

closed 2018-11-01 by zhouq123