realsenseai/librealsense #2630: Playback Start Time

philipjames44, 2018-10-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |   MacOS  | 
| Platform                                 | PC  |
| Language                            |  C++                          | 

I've been using the playback tutorial to successfully stream video from a rosbag frame by frame, however I would like to specify a specific start time or frame and haven't found documentation for such as of yet. Is there a convenient way to specify start time of a stream from a rosbag?

MartyG-RealSense, 2018-10-28:
Other users have wished for this feature, describing it in terms of accessing bag frames by index number.  In SDK release 2.16.1 a function called set_real_time that had previously had problems was bug-fixed, making it possible to fetch frames one frameset at a time.  A user subsequently created a script for browsing a bag frame by frame.

https://github.com/IntelRealSense/librealsense/issues/1579#issuecomment-430163908

philipjames44, 2018-10-29:
@MartyG-RealSense I was able to replicate the reading of the frames frame by frame from a frameset. However, I do not see how this can translate to accessing frames by index. Do you mean to say this is not implemented as of yet?

philipjames44, 2018-10-29:
Sure. Thanks for your help @MartyG-RealSense ! I found this documentation, so at the moment I'm using the seek method to select a starting point based off of the nanosecond of the frame desired: http://docs.ros.org/kinetic/api/librealsense2/html/classrs2_1_1playback.html#ae1ad770b827a5794a83b37826ffdcb2a

dorodnic, 2018-10-29:
Thanks @MartyG-RealSense 

We decided not to implement "seek to frame index" for now. Once realtime / non-realtime bug was resolved, you can always seek to start and quickly `wait_for_frames` in a loop until the frame number you need. We didn't put it in API just yet, because frame index is less defined compared to global time. (the frame number of the frameset? of the depth frame? it is not 100% obvious) 

@philipjames44 let us know if you need any help. Also the [record-playback](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback) example has some information, as well as (somewhat hidden) [src/media](https://github.com/IntelRealSense/librealsense/tree/master/src/media)

RealSense-Customer-Engineering, 2018-11-05:
[Realsense Customer Engineering Team Comment]
Hi philipjames44,

Do you still need any help given what have been explained above?

Thank you!

philipjames44, 2018-11-06:
I believe this issue is resolved. Thanks!

closed 2018-11-06 by philipjames44