realsenseai/librealsense #2632: [C++] Error compiling librealsense/wrappers/opencv/dnn

HippoEug, 2018-10-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

# Introduction
Hello everybody! I wrote [this](https://github.com/IntelRealSense/librealsense/issues/2584) question couple days ago, `Is depth.get_distance(x,y) a time consuming function on C++?`

Dorodnic mentioned `depth.get_distance` is not recommended in my use case, instead "multiply the raw data by the value of RS2_OPTION_DEPTH_UNITS option", along with [this OpenCV example](https://github.com/IntelRealSense/librealsense/blob/c3c758d18c585a237bb5b635927797aa69996391/wrappers/opencv/cv-helpers.hpp#L43-L56).

To apply what he mentioned, I tried compiling `librealsense/wrappers/opencv/dnn` [(quick link)](https://github.com/IntelRealSense/librealsense/tree/c3c758d18c585a237bb5b635927797aa69996391/wrappers/opencv/dnn).

Similar Issue: [#1634 Errors compiling with OpenCV examples](https://github.com/IntelRealSense/librealsense/issues/1634)

# Error
#### `This is the error I got on compilation..`
```
OpenCV(3.4.3) Error: Unspecified error (FAILED: fs.is_open(). Can't open "MobileNetSSD_deploy.prototxt") in cv::dnn::ReadProtoFromTextFile, file c:\build\3_4_winpack-build-win64-vc15\opencv\modules\dnn\src\caffe\caffe_io.cpp, line 1121
OpenCV: terminate handler is called! The last OpenCV error is:
OpenCV(3.4.3) Error: Unspecified error (FAILED: fs.is_open(). Can't open "MobileNetSSD_deploy.prototxt") in cv::dnn::ReadProtoFromTextFile, file c:\build\3_4_winpack-build-win64-vc15\opencv\modules\dnn\src\caffe\caffe_io.cpp, line 1121
```
[image]

#### `Here is how the test project looks:`
[image]
[image]

There are no build errors whatsoever. Have never used OpenCV much, hence I am not sure what that error really is. Does anyone know if I need to add anymore files or something to fix this problem? 

Gracias! 🐱🙏😸

dorodnic, 2018-10-29:
Hi @HippoEug 

The simplest way to get started with OpenCV right now is actually using newly published installer - [Intel.RealSense.SDK-2.16.2.304.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.2/Intel.RealSense.SDK-2.16.2.304.exe)
Based on your and others feedback we tried to package everything up neatly, so you will get a cleanly compiling opencv examples as part of samples.sln. (there are still couple of things missing, but `rs-dnn` is included). 

Regarding the runtime error you are getting - this example relies on MobileNet neural-network that we download during CMake. In particular you are missing the `MobileNetSSD_deploy.prototxt` file (or perhaps it is in wrong location?). 

As for faster conversion to meters, you don't actually **require** `opencv` to do this. All you need is to fetch the `depth_scale`:
```cpp
auto depth_scale = pipe.get_active_profile()
        .get_device()
        .first<depth_sensor>()
        .get_depth_scale();
```
And then multiply every pixel by this value, something like:
```cpp
uint16_t* pixels = (uint16_t*)depth_frame.get_frame_data();
int w = depth_frame.get_width();
int h = depth_frame.get_height()
for (int j = 0; j < h; j++)
for (int i = 0; i < w; i++)
        float dist_in_meters = pixels[i * w + j] * depth_scale; // should give the same result depth_frame.get_distance(i, j) but mach faster
```
I do, however, encourage to try the OpenCV examples - OpenCV is awesome!

dorodnic, 2018-11-08:
Hi @HippoEug 
Did you try the new installer sample apps?

HippoEug, 2018-11-08:
Hi @dorodnic 

The installer looks good, with the 'glfw-imgui.props' especially. Since I have VS2017, I am unable to build the solution. Must I download VS2015?

dorodnic, 2018-11-08:
I think there is an option, called something like "Retarget Solution" that would prompt a window like this:
[image]

HippoEug, 2018-11-09:
Hi @dorodnic 

### Danke
Thanks for the suggestion! Re-targeted to Windows SDK Version: 10.17134.0 for VS2017 and it works!

### Feedback
First of all, the installer is amazing! The libraries are great and all the examples just work. On the previous version (.16.1), I was experiencing some compilation issues with OpenCV/Imgui/GLFW libraries here and there.

Also, almost all of the existing examples are still the same, but `rs-align.cpp` has better performance for some reason, even though the code on the surface is still the same. I got much better FPS, though I don't know the exact values.

### CV-examples
These examples work too after a simple re-target. Although the FPS is really bad, could be my slow laptop. Here are 2 DNN outputs:

[image]
[image]

dorodnic, 2018-11-09:
Great,

Regarding tools, I don't thing their source-code needs to be part of the project. The point of samples is more functionality but also much more code. Unless you wish to debug and extend the Viewer, we don't really expect you to go into its code, as it is quite complicated. We do distribute some of the tools in form of binaries, but `rs-convert` might still be missing. 

The DNN is supposed to be kinda slow - the detection part is completely of-the-shelf. There are many solutions to accelerate object detection, but we want to focus on the depth-sensing technology.

HippoEug, 2018-11-09:
I see, thank you. I don't think there is anything more. Closing this issue 😄

closed 2018-11-09 by HippoEug