realsenseai/librealsense #2633: How to decrease an error including output depth value

fujiimakoto, 2018-10-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version | Win10 | 
| Platform                                 | PC |
| SDK Version                            | RealSense SDK 2.0 version 2.16.1.251 | 
| Language                            | python | 
| Segment			| others | 

### Issue Description

Hello, all.

I'm writing my python application to measure the distance from Realsense to an obstacle.
When my application is run,  output depth values have an error.
So, I read 'Understand theoretical limit' in this pdf, I push Realsense and an obstacle closer and closer together to decrease an error.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

However, an error doesn't decrease, and keeps constant(about 10mm).
Do you have any solutions to decrease an error?

Best regards,

agrunnet, 2018-10-29:
Can you give us more details about what distances you have and what object you are looking at? How are you determining the ground truth also?

RealSense-Customer-Engineering, 2018-11-05:
[Realsense Customer Engineering Team Comment]
Hi fujiimakoto ,

Would you like to first validate your camera using depth quality tool https://github.com/IntelRealSense/librealsense/releases?
And you can refer to this C-example for measuring depth: https://github.com/IntelRealSense/librealsense/tree/master/examples/C/distance

Hope these can be helpful.

Thank you!

fujiimakoto, 2018-11-07:
Hi Realsense Customer Engineering Team,

Using depth quality tool and turning off auto_exposure, these error distances is less than or equal to 1mm.
Thank you for your cooperation.

closed 2018-11-07 by fujiimakoto