realsenseai/librealsense #2634: How to convert cv::mat to rs::frame?

Liuyu010, 2018-10-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.09.02.0) | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  {2.13.0 }                          | 
| Language                            |  {C++/opencv }                          | 
| Segment			|  VR/AR }                          | 

### Issue Description
Hello,

I have already seen this question in the realsense community but didn't get the answer I want.

In rs_align, I have converted the rs2:: frame to cv:: Mat for using the joint bilateral filter provided by OpenCV to improve the quality of the depth map. The depth map of the processed mat type that is obtained cannot be used to call _remove_background(rs2::video_frame& other, const **rs::frame & depth_frame**, float depth_scale, float clipping_dist)_.

So, I just want to know how to convert **cv:: Mat to  rs2:: frame**. If you can help me, I would be very grateful.

Thank you!

dorodnic, 2018-10-29:
Hi @Liuyu010 
Frame is not exactly equivalent to cv::Mat, as it carries a lot of camera specific information (link to the sensor, metadata, link to calibration, etc...) That's why there is no **easy** cv-->frame API.
It is possible, of course. I made a modified version of im-show example demonstrating how to go about it: 

```cpp
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API

int main(int argc, char * argv[]) try
{
    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start();

    using namespace cv;
    const auto window_name = "Display Image";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    // Create custom processing block and its output queue:
    rs2::frame_queue q;
    rs2::processing_block pb(
        [](rs2::frame f, rs2::frame_source& src)
    {
        // For each input frame f, do:

        const int w = f.as<rs2::video_frame>().get_width();
        const int h = f.as<rs2::video_frame>().get_height();

        // frame --> cv
        Mat image(Size(w, h), CV_16UC1, (void*)f.get_data(), Mat::AUTO_STEP);
        // do some (silly) processing
        cv::GaussianBlur(image, image, cv::Size(0, 0), 3);

        // Allocate new frame. Copy all missing data from f.
        // This assumes the output is same resolution and format
        // if not true, need to specify parameters explicitly
        auto res = src.allocate_video_frame(f.get_profile(), f);

        // copy from cv --> frame
        memcpy((void*)res.get_data(), image.data, w * h * 2);

        // Send the resulting frame to the output queue
        src.frame_ready(res);
    });
    pb.start(q); // Bind output of the processing block to be enqueued into the queue

    while (waitKey(1) < 0 && cvGetWindowHandle(window_name))
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        rs2::frame depth = data.get_depth_frame();

        // Send depth frame for processing
        pb.invoke(depth);
        // Wait for results
        depth = q.wait_for_frame();

        // Continue with the new frame
        depth = depth.apply_filter(color_map);

        // Query frame size (width and height)
        const int w = depth.as<rs2::video_frame>().get_width();
        const int h = depth.as<rs2::video_frame>().get_height();

        // Create OpenCV matrix of size (w,h) from the colorized depth data
        Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

Liuyu010, 2018-10-30:
Hi，dorodnic
Thank you for your reply! Your advice sure makes sense and I have tried to use it in my project. But there are still two problems that bother me.

1. In the processing_block pb(), not only do I want to pass depth_map into it but also include color_map. When I tried to modify it, I was often reminded that "this function will not be counted as accepting two parameters".How to transfer two parameters to processing_block pb()?
2. All my project are based on **rs-align**. So, I tried to use the method you provided in rs_align to process depth_map with **OpenCV** to see if it could be used as a way to improve **background segmentation**, but it doesn't work. Didi do something wrong?
   - Here is the wrong result.
 
[image]

   - Here is my code.
https://github.com/vivedu/Third-Person-VR/blob/JointBilateralFilter/segmentation.cpp

Finally, I would like to ask you whether there is a good background segmentation program or a good idea on the D400s realsense camera. Although I have tried all the filters you provided in rs_align(like deci_filter、spat_filter、temp_filter、hole_filter etc.), the result is still a serious situation of edge leakage. Looking forward to your reply.

dorodnic, 2018-10-30:
Hi @Liuyu010 

To pass two or more frames, you need to bundle them in a frameset. See an example here: [rs-measure.cpp:191-233](https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp#L191-L233)

Third-Person-VR seem to be a private repository, I couldn't access the code. 

In general, I think that good background removal would leverage both color and depth data. In `rs-grabcuts` I tried to feed depth data as initial guess to purely RGB based graph-cuts background removal. This can show decent results, but performance is abysmal.
I wanted to try a hybrid approach for a while now, but couldn't make it work yet. 

It is obvious that depth data **helps** in the problem of background removal, but in terms of depth sensing solutions, we are not yet in a place where you could just take the raw depth map and use it as a threshold. Main issue is that most depth cameras available today are composed from several sensors, each at slightly different location. This means that you will always going to get pixels that one sensor sees but the second does not, resulting in black pixels. 

That said, I'm a software engineer, not a computer vision expert, so I'm sure there are people here who have better ideas.

P.S. I tried to fine-tune the example a bit for the D415 and came up with two suggestions:
1. You can fairly easily enable basic post-processing to get less holes:
```cpp
rs2::temporal_filter temp;
rs2::hole_filling_filter hfill;

while (app) // Application still alive?
{
    // Using the align object, we block the application until a frameset is available
    rs2::frameset frameset = pipe.wait_for_frames().apply_filter(temp).apply_filter(hfill);
```
2. The D415 can deliver imperfect RGB from the left IR sensor (potentially reducing calibration artifacts)
You can try it and compare to the results from RGB sensor: 
```cpp
    rs2::pipeline pipe;

    rs2::config config;
    config.enable_stream(RS2_STREAM_DEPTH);
    config.enable_stream(RS2_STREAM_INFRARED, RS2_FORMAT_RGB8);
    rs2::pipeline_profile profile = pipe.start(config);

    auto sensor = profile.get_device().query_sensors().front();
    sensor.set_option(RS2_OPTION_VISUAL_PRESET, 6.0);
    sensor.set_option(RS2_OPTION_EXPOSURE, 50000.0);
    sensor.set_option(RS2_OPTION_LASER_POWER, sensor.get_option_range(RS2_OPTION_LASER_POWER).max);
```

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Hi @Liuyu010, If below answer have solved your question, we'd like to close this ticket. Thanks!

Liuyu010, 2018-11-06:
Hi，dorodnic
Sorry for reply so late! 
I have tried what you advised last week, but it seems that I made the project more Chaotic and can't get the features I want(Maybe because of my unfamiliarity with developing on Intel RealSense).
In rs_align, all the features I want is that I can use the color_map and depth_map fetched from D415/D435 to add the Joint Bilateral Filter on the depth_map(This is a method I think can improve the edge effect of the map very well). But I am still stuck in unknowing how to use the depth_image processed by OpenCV to call  remove_background().
My current problem is the same as the two problems mentioned earlier. I have tried the pb() and changing the format of the input of the depth map of remove_background(), but all failed,
Here is my code. Sincerely hope to get your help!

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp>
#include "window.hpp"
#include <imgui.h>
#include "imgui_impl_glfw.h"

#include <sstream>
#include <iostream>
#include <fstream>
#include <algorithm>
#include <cstring>

#include <opencv2/opencv.hpp>
#include <ximgproc.hpp>

void render_slider(rect location, float& clipping_dist);
void remove_background(rs2::video_frame& other_frame, const rs2::depth_frame& depth_frame, float depth_scale, float clipping_dist);
float get_depth_scale(rs2::device dev);
rs2_stream find_stream_to_align(const std::vector<rs2::stream_profile>& streams);
bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev);

int main(int argc, char * argv[]) try
{
    // Create and initialize GUI related objects
    window app(1280, 720, "CPP - Align Example"); // Simple window handling
    ImGui_ImplGlfw_Init(app, false);      // ImGui library intializition
    rs2::colorizer c;                          // Helper to colorize depth images
    texture renderer;                     // Helper for renderig images

    // Create a pipeline to easily configure and start the camera
    rs2::pipeline pipe;
    //Calling pipeline's start() without any additional parameters will start the first device
    // with its default streams.
    //The start function returns the pipeline profile which the pipeline used to start the device
    rs2::pipeline_profile profile = pipe.start();

    // Each depth camera might have different units for depth pixels, so we get it here
    // Using the pipeline's profile, we can retrieve the device that the pipeline uses
    float depth_scale = get_depth_scale(profile.get_device());

    //Pipeline could choose a device that does not have a color stream
    //If there is no color stream, choose to align depth to another stream
    rs2_stream align_to = find_stream_to_align(profile.get_streams());

    // Create a rs2::align object.
    // rs2::align allows us to perform alignment of depth frames to others frames
    //The "align_to" is the stream type to which we plan to align depth frames.
    rs2::align align(align_to);

    // Define a variable for controlling the distance to clip
    float depth_clipping_distance = 1.f;

	// Create custom processing block and its output queue:
	rs2::frame_queue q;
	rs2::processing_block pb(
		[](rs2::frame f, rs2::frame_source& src)
		//[](rs2::frame f, rs2::frame color,rs2::frame_source& src)
	{

		// For each input frame f, do:
[... 236 lines cut ...]

    if(!depth_stream_found)
        throw std::runtime_error("No Depth stream available");

    if (align_to == RS2_STREAM_ANY)
        throw std::runtime_error("No stream found to align with Depth");

    return align_to;
}

bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev)
{
    for (auto&& sp : prev)
    {
        //If previous profile is in current (maybe just added another)
        auto itr = std::find_if(std::begin(current), std::end(current), [&sp](const rs2::stream_profile& current_sp) { return sp.unique_id() == current_sp.unique_id(); });
        if (itr == std::end(current)) //If it previous stream wasn't found in current
        {
            return true;
        }
    }
    return false;
}

```

RealSense-Customer-Engineering, 2018-11-13:
[Realsense Customer Engineering Team Comment]
Hi @Liuyu010 ,

from your posts, your question is "how to use the depth_image processed by OpenCV to call remove_background()",  could you reference following modified rs-align.cpp(librealsense/examples/align/rs-align.cpp):

// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp>
#include "../example.hpp"
#include <imgui.h>
#include "imgui_impl_glfw.h"

#include <sstream>
#include <iostream>
#include <fstream>
#include <algorithm>
#include <cstring>

#include <opencv2/opencv.hpp>  // Include OpenCV API
using namespace cv;

void render_slider(rect location, float& clipping_dist);
void remove_background(rs2::video_frame& other, const rs2::depth_frame& depth_frame, float depth_scale, float clipping_dist);
float get_depth_scale(rs2::device dev);
rs2_stream find_stream_to_align(const std::vector<rs2::stream_profile>& streams);
bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev);

int main(int argc, char * argv[]) try
{
    // Create and initialize GUI related objects
    window app(1280, 720, "CPP - Align Example"); // Simple window handling
    ImGui_ImplGlfw_Init(app, false);      // ImGui library intializition
    rs2::colorizer c;                          // Helper to colorize depth images
    texture renderer;                    // Helper for renderig images

    // Create a pipeline to easily configure and start the camera
    rs2::pipeline pipe;
    //Calling pipeline's start() without any additional parameters will start the first device
    // with its default streams.
    //The start function returns the pipeline profile which the pipeline used to start the device
    rs2::pipeline_profile profile = pipe.start();

    // Each depth camera might have different units for depth pixels, so we get it here
    // Using the pipeline's profile, we can retrieve the device that the pipeline uses
    float depth_scale = get_depth_scale(profile.get_device());

    //Pipeline could choose a device that does not have a color stream
    //If there is no color stream, choose to align depth to another stream
    rs2_stream align_to = find_stream_to_align(profile.get_streams());

    // Create a rs2::align object.
    // rs2::align allows us to perform alignment of depth frames to others frames
    //The "align_to" is the stream type to which we plan to align depth frames.
    rs2::align align(align_to);

    // Define a variable for controlling the distance to clip
    float depth_clipping_distance = 1.f;

// Create custom processing block and its output queue:
    rs2::frame_queue q;
    rs2::processing_block pb(
        [](rs2::frame f, rs2::frame_source& src)
    {
        // For each input frame f, do:

        const int w = f.as<rs2::video_frame>().get_width();
        const int h = f.as<rs2::video_frame>().get_height();

        // frame --> cv
[... 252 lines cut ...]
elseif(COMPILER_SUPPORTS_CXX0X)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
endif()

if(BUILD_GRAPHICAL_EXAMPLES)
    #align
    add_executable(rs-align rs-align.cpp ../../third-party/imgui/imgui.cpp ../../third-party/imgui/imgui_draw.cpp ../../third-party/imgui/imgui_impl_glfw.cpp)
    #    target_link_libraries(rs-align ${DEPENDENCIES})
    target_link_libraries(rs-align ${DEPENDENCIES} ${OpenCV_LIBS})
    include_directories(rs-align ../../common ../../third-party/imgui)
    set_target_properties (rs-align PROPERTIES
        FOLDER Examples
    )

    install(
        TARGETS

        rs-align

        RUNTIME DESTINATION
        ${CMAKE_INSTALL_PREFIX}/bin
    )
endif()

we'd like to hear from you later.

Liuyu010, 2018-11-14:
Thank you for your patience.
I have tried what you suggested, but the output window is still dark.

[image]

After a step-by-step debugging, the preliminary judgment is `aligned_depth_frame = q.wait_for_frame()` cannot assign depth_frame from the queue to aligned_depth_frame.

Have you encountered any similar problems here? Looking forward to your reply.

RealSense-Customer-Engineering, 2018-11-16:
[Realsense Customer Engineering Team Comment]
Hi @Liuyu010,

we need to dig more into this issue. Thanks for your investigation and patience.

Liuyu010, 2018-11-16:
Hi @RealSense-Customer-Engineering 
According to the [measure.cpp:191-233](https://github.com/IntelRealSense/librealsense/blob/c1be270c96ef07faa7ceecf4117890ce06af65a8/examples/measure/rs-measure.cpp#L191-L233), when I tend to pass in two parameters(color_frame and depth_frame) at the same time to **pb()** , there are still a lot of syntax errors. Can you provide a more intuitive way? 

In addition, your company recently released the D435i camera, will the camera have a greater effect on the rs_align? Is there any test results?

Thank you.

Liuyu010, 2018-11-22:
Hi @dorodnic ，
Can you help me about transferring two parameters(color_frame and depth_frame) to pb()?
I have tried much times, but still trapped. This is the last question, sincerely hope to get your help.

RealSense-Customer-Engineering, 2018-12-03:
[Realsense Customer Engineering Team Comment]
Hi @Liuyu010 ，

hope the following new rs-align.cpp meet your requirement:

// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp>
#include "../example.hpp"
#include <imgui.h>
#include "imgui_impl_glfw.h"

#include <sstream>
#include <iostream>
#include <fstream>
#include <algorithm>
#include <cstring>

#include <opencv2/opencv.hpp>  // Include OpenCV API
using namespace cv;

void render_slider(rect location, float& clipping_dist);
void remove_background(rs2::video_frame& other, const rs2::depth_frame& depth_frame, float depth_scale, float clipping_dist);
float get_depth_scale(rs2::device dev);
rs2_stream find_stream_to_align(const std::vector<rs2::stream_profile>& streams);
bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev);

int main(int argc, char * argv[]) try
{
    // Create and initialize GUI related objects
    window app(1280, 720, "CPP - Align Example"); // Simple window handling
    ImGui_ImplGlfw_Init(app, false);      // ImGui library intializition
    rs2::colorizer c;                          // Helper to colorize depth images
    texture renderer;                    // Helper for renderig images

    // Create a pipeline to easily configure and start the camera
    rs2::pipeline pipe;
    //Calling pipeline's start() without any additional parameters will start the first device
    // with its default streams.
    //The start function returns the pipeline profile which the pipeline used to start the device
    rs2::pipeline_profile profile = pipe.start();

    // Each depth camera might have different units for depth pixels, so we get it here
    // Using the pipeline's profile, we can retrieve the device that the pipeline uses
    float depth_scale = get_depth_scale(profile.get_device());

    //Pipeline could choose a device that does not have a color stream
    //If there is no color stream, choose to align depth to another stream
    rs2_stream align_to = find_stream_to_align(profile.get_streams());

    // Create a rs2::align object.
    // rs2::align allows us to perform alignment of depth frames to others frames
    //The "align_to" is the stream type to which we plan to align depth frames.
    rs2::align align(align_to);

    // Define a variable for controlling the distance to clip
    float depth_clipping_distance = 1.f;

// Create custom processing block and its output queue:
    rs2::frame_queue q;
    rs2::processing_block pb(
        [&](rs2::frameset data, rs2::frame_source& src)
    {

        rs2::frame f = data.get_depth_frame();
        // For each input frame f, do:

        const int w = f.as<rs2::video_frame>().get_width();
        const int h = f.as<rs2::video_frame>().get_height();
[... 214 lines cut ...]

    if(!depth_stream_found)
        throw std::runtime_error("No Depth stream available");

    if (align_to == RS2_STREAM_ANY)
        throw std::runtime_error("No stream found to align with Depth");

    return align_to;
}

bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev)
{
    for (auto&& sp : prev)
    {
        //If previous profile is in current (maybe just added another)
        auto itr = std::find_if(std::begin(current), std::end(current), [&sp](const rs2::stream_profile& current_sp) { return sp.unique_id() == current_sp.unique_id(); });
        if (itr == std::end(current)) //If it previous stream wasn't found in current
        {
            return true;
        }
    }
    return false;
}

and our expert @dorodnic had been pointed out that you could pass frameset(contain color_frame and depth_frame and ir_frame...) to the pb.

RealSense-Customer-Engineering, 2018-12-03:
[Realsense Customer Engineering Team Comment]
Hi @Liuyu010,

we had been validated the latest modified rs-align.cpp, and it works find as expected.

we'd like to close this ticket if it could solved your question.

we'd like to hear from you again.

Thanks!

Best regards

RealSense-Customer-Engineering, 2018-12-10:
[Realsense Customer Engineering Team Comment]
@Liuyu010 Did you get any chance to try what we suggested to get this through? Please update. Thanks!

RealSense-Customer-Engineering, 2018-12-14:
[Realsense Customer Engineering Team Comment]
Hi @Liuyu010,

any progress with this issue? we'd like to get a positive response from you.

Thanks.

Liuyu010, 2018-12-21:
Sorry for the later response.
We have successfully processed RGB images and depth images simultaneously to get a better depth map in pb(). And when we tend to use Background_romoval by this processed depth map, the background's depth_value of the map seems to be very confusing（As shown below.）. So, we are going to give up this plan, Thank you for your continued reply.
[image]

[image]

[image]

RealSense-Customer-Engineering, 2018-12-24:
[Realsense Customer Engineering Team Comment]
@Liuyu010 Sorry I'm not very clear about what you mentioned about "the map seems to be very confusing". Could you please explain more about it?

Liuyu010, 2018-12-26:
Ummm...it means that the depth map's precision does not support the background removal very well. We are looking for other possible solutions.
Thank you for your sincere help.

RealSense-Customer-Engineering, 2018-12-26:
[Realsense Customer Engineering Team Comment]
@liuyu010 Thanks for the feedback! We have one such example in the link https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts  You can have a try. Thanks!

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
@liuyu010 Thanks for your using Realsense camera! Close this ticket. If you still have other questions, please feel free to open new ticket.

closed 2019-01-03 by RealSense-Customer-Engineering