realsenseai/librealsense #2635: How to playback ".bag"-files properly in C#/Unity

NeuroRanger, 2018-10-29:
I'm using the "wrappers/unity"-C#-environment of the current SDK-Version "2.16".

What is working for me is creating an "RsDevice"-instance, set the mode to "playback", set the filename and enable it.
I'd like to have more control over the playback-process, e.g. get the total framecount, wind to a specific frame and use the data, start/stop playback, ...
Do I have to go deeper into using the "Pipeline", own WorkerThread, etc. or is it maybe possible to realize those functions with "RsDevice" respectively "RsFrameProvider"?

I also found this "rs2::playback"-Function:
https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md
This propably is C/C++-specific. I guess, the namespace "rs2" is "Intel.RealSense" in C#, but I can't find a "Playback"-Class or something similar there.

Would be great to get some tips on working with .bag-files in C#/Unity.
Thanks & Greetings.

----------------------------------------------------------------------------------------------------

| Infos                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                       |
| Firmware Version | () |
| Operating System & Version | Win 10 |
| Platform | PC |
| SDK Version | 2.16 |
| Language | C/C#/unity |
| Segment | others |

### Issue Description
<Playback of .bag-Files in C#/Unity>

NeuroRanger, 2018-11-01:
Thanks, but the "Context"-Object doesn't have a Method called "AddDevice".
The rest is able to compile, I can get Sensor-Infos and switch the Status from "Pause" to "Playing". But I don't receive any NewFrame-Events when using

sensor.Start(OnNewSample);
...
  void OnNewSample(Frame frame)
  {
    Debug.Log("New Sample");
  }

Are there maybe some examples on how to use the "PlaybackDevice"-Class?

NeuroRanger, 2018-11-03:
Well, I tested a bit with "RsDevice" and "PlaybackDevice".
Here's a little MonoBehaviour-Script, that should at least be able to Start and Stop the Playback and eventually receive Frames:
### Test01
```
using System.Collections;
using UnityEngine;
using Intel.RealSense;

public class Test01 : MonoBehaviour
{
  RsDevice       rsdevice;
  PlaybackDevice pbDev   ;
  Sensor         sensor  ;

  void OnEnable()
  {
    rsdevice = GetComponent<RsDevice>();
    if (rsdevice != null) Debug.Log("Device found");

    rsdevice.DeviceConfiguration.mode         = RsConfiguration.Mode.Playback;
    rsdevice.DeviceConfiguration.PlaybackFile = "D:/developement/3d/realsense/data/20181018_213640.bag";

    rsdevice.enabled = true;

    StartCoroutine(Init());
  }
  IEnumerator Init()
  {
    yield return new WaitForSeconds(1.0f);

    pbDev = PlaybackDevice.FromDevice(rsdevice.ActiveProfile.Device);

    Debug.Log("Status at start = " + pbDev.Status);

    sensor = pbDev.Sensors[0];

    for(int i = 0; i < sensor.StreamProfiles.Count; i++)
    {
      sensor.Open(sensor.StreamProfiles[i]);
    }

    sensor.Start(OnNewSample);

    pbDev.Pause();

    rsdevice.OnNewSample += OnNewSample;

    Debug.Log("Initialized; Device-Status = " + pbDev.Status);

    yield break;
  }
  private void OnGUI()
  {
    if(pbDev != null)
    {
      if(GUI.Button(new Rect(50, 50, 50, 30), "Start"))
      {
        Debug.Log("Start pressed");

        pbDev.Resume();
        Debug.Log("Status after Resume() = " + pbDev.Status);
      }
      if(GUI.Button(new Rect(50, 100, 50, 30), "Stop"))
      {
        Debug.Log("Stop pressed");

        pbDev.Pause();
        Debug.Log("Status after Pause() = " + pbDev.Status);
      }
    }
  }
  void OnNewSample(Frame frame)
  {
    Debug.Log("New Sample: " + frame.Number);
  }
}
```

The Console-Output in Unity looks like this:
[image]

There's an Access-Violation:

> Exception: Access violation - no RTTI data!
rs2_playback_device_resume(device:000000005935D6C0)
Intel.RealSense.Helpers+ErrorMarshaler.MarshalNativeToManaged (IntPtr pNativeData)
Intel.RealSense.PlaybackDevice.Resume ()
Test01.OnGUI () (at Assets/Scripts/Test01.cs:58)

Should it work somehow like this? What am I missing or doing wrong?

RealSense-Customer-Engineering, 2018-12-12:
[Realsense Customer Engineering Team Comment]
Ticket being closed due to inactivity for 30+ days

closed 2018-12-12 by RealSense-Customer-Engineering

NeuroRanger, 2018-12-14:
There is no solution yet.
The only answer didn't work,.
I'd like to process recorded realsense-data in .bag-files inside Unity.

NeuroRanger, 2018-12-14:
So, could someone please open this Topic again?

iMilk, 2019-01-10:
How do I jump to the next frame with a button when I using WindowsFom(c#)?

Dealerpriest, 2019-03-01:
I have the same error: ```Access violation - no RTTI data!```

What could be the reason?

millsrandy, 2019-06-06:
Did you guys ever get this to work?  I'm trying to do the same thing in Unity with pose data from the T265.  I need to get the .bag data as frame data so I can export an fbx animation of my camera to C4D

julioobelleiro, 2019-09-16:
any updates on this? also trying to play .bag files (depth and rgb) from Unity... any help appreciated!

Dealerpriest, 2019-09-16:
I am not experiencing this issue anymore. And I can't really assist in anything specific measures to take.
But perhaps upgrade to latest version of the unity package? At the time of writing this it's [v2.28.0](https://github.com/IntelRealSense/librealsense/releases/download/v2.28.0/Intel.RealSense.unitypackage)
Also. Are you able to playback files in the unity example scene? If so. It might be related to your code.

julioobelleiro, 2019-09-16:
ah thanks, it might be he version of the sdk then. will update and check, thanks for the quick reply