realsenseai/librealsense #2638: May be I can scanner by Intel D435 Depth with an object ?

NguyenThaiHoc1, 2018-10-30:
I need to scan a face of person 

Can i do that ? and if you talk yes, How I scan ? 

I try to LIPScan3D application But it did not working (I can try to install application on my computer but not work. I see more detail in their website, they said that application is not complete !!)

I hope you support me about this problem 
thanks you
Thai Hoc,

HippoEug, 2018-10-30:
Hi, which camera are you using? Have you tried using the Intel Realsense Viewer?

You can capture a Pointcloud (PLY) using the default Realsense Viewer.
1. Turn both modules on as seen here.
[image]

2. Change to 3D mode and you can see the Pointcloud and manipulate it in real time.
[image]

3. Press the Save icon on the top right, under the 2D button. You can import this file into Meshlab for example to see the pointcloud.

NguyenThaiHoc1, 2018-10-30:
I want to have image like link below, How can i get it from camera ? 

https://1drv.ms/u/s!AivWk88jn2vKgUeKkUUOWaUOp0XP

HippoEug, 2018-10-30:
Hi,

Here is the image from the above link:
[image]

Here is the output from the Realsense Z16 depth camera. I am not sure if this is what you want..

NguyenThaiHoc1, 2018-10-30:
hi @HippoEug 

Thanks you for support me !

another question

How can i combine RGB image and  Depth Image Z16 ?

MartyG-RealSense, 2018-10-30:
It is possible to align RGB and depth in the 3D mode of the RealSense Viewer using the procedure in the link below.

https://communities.intel.com/message/567566#567566

RealSense-Customer-Engineering, 2018-11-02:
[Realsense Customer Engineering Team Comment]
Hi NguyenThaiHoc1 ,

Has your question been resolved by referring below link? If yes, we'd like to close this ticket. Thanks!

closed 2018-11-14 by RealSense-Customer-Engineering