realsenseai/librealsense #2647: Asynchronous callback for librealsense2

yashmanian, 2018-10-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |  Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.15.0.36-generic                                         | 
| Platform                                 | PC  |
| Language                            |  C++ | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am looking for a way to use the Intel RealSense asynchronously, where I can define a callback which will return frames once the camera has them. Currently, the only way I could find to get frames is to use wait_for_frames() or poll_for_frames(), both of which are a part of the pipeline.
Making the assumption that the pipeline runs in a single thread, is there a way to get frames asynchronously,  so that the frames are updated separately from the main thread?
The only document I could find mentioning asynchronous frame buffering is:https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0
It would be great if someone could point me in the direction of more descriptive documentation.

dorodnic, 2018-10-31:
Please take a look at `sensor-control` example, specifically [api_how_to.h:472](https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L472)

yashmanian, 2018-10-31:
Thank you. 
Looking at the code, am I correct in making the assumption that the void run() function inside the frame viewer, is what gets called every time frame f is updated inside of sensor.start() in a separate thread? 
And that I only need to call sensor.start() to start said thread, after which it will continue to run asynchronously?

dorodnic, 2018-10-31:
I'm not sure what run function you are referring to...
`[&](rs2::frame f) { display(f); }` will get called every time frame f is updated. 
You can also pass a pointer to regular function, or an `std::function` or anything with `operator()(frame f)`.
You only need to call `sensor.start()`, it returns instantly and until you call `stop` you will get notified on all frames from that sensor. The callback will get invoked not on the main thread, but some other thread. (on Linux librealsense manually creates and manages thread per device, on Windows you actually get frames asynchronously so the callback will run on the Media Foundation thread).
The point is, except some basic pixel unpacking (that you control by specifying RS2_FORMAT), no extra processing or context switching will occur.

dorodnic, 2018-10-31:
Configuring sensors is a bit more verbose (you need to manually pick one or more out of the stream profiles returned by `get_stream_profiles()`, there are no wildcards), but it maps much more accurately to hardware. 

We discussed adding callback option to `pipeline.start`, to get synchronized frame-sets asynchronously. Maybe at some time in the future. Right now, you can still achieve this effect using the strangely named `asynchronous_syncher` class. Just direct the two sensors you want to synchronize to this object, and it will output frame-sets via a callback of its own.

yashmanian, 2018-10-31:
Thank you. That is helpful. I will be closing the issue now

closed 2018-10-31 by yashmanian