realsenseai/librealsense #2648: Unity plugin causes Blue Screens every time played and paused

sepehr-laal, 2018-10-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win 10 - 1803  | 
| Platform                                 | PC (Unity3D)  |
| SDK Version                            |  2.16.1 | 
| Language                            |  unity }                          | 

### Issue Description

Play the sample "Depth and IR texture" scene, let it go on for a couple of seconds and stop the editor. Wait a couple of minutes in the editor and you get a blue screen with "IRQL NOT LESS OR EQUAL".

The workaround is to completely close the editor after every stop! which means a complete restart of Unity editor every time I want to play the scene. I am not an expert in drivers but this seems like something is not cleaned up correctly.

RealSense-Customer-Engineering, 2018-11-05:
[Realsense Customer Engineering Team Comment]
Hi sepehr-laal,

I tried with Unity 2018.2.13f1 with RealSense SDK 2.16.1 Unity wrapper off Windows 10 Version 1709 Build 16299.726, and don't have issue when strting/stopping TexturesDepthAndInfrared.unity sample scene, neither for StartHere.unity sample scene.

Can you please help clarify the issue you have, and take some screenshots?

Thanks!

sepehr-laal, 2018-11-06:
I can't take a screenshot of the blue screen. I am on Unity 2018.2.14f1.

This is how I reproduce it:

1. Make an empty scene
2. Drag drop RsDevice prefab into it
3. Set it up to stream depth, IR or RGB (does not matter as long as it's streaming something)
4. Play
5. Stop
6. Wait around for 10 to 15 minutes
7. Blue Screen.

The issue is that Unity Editor needs to be closed an reopened every time a scene with a RsDevice prefab in it is played. If you leave the Unity Editor open, it will result in a blue screen after 10 to 15 minutes (I am assuming this is when C# GC hits something dangling).

sepehr-laal, 2018-12-10:
I just wanted to report back that, since the time I posted this issue both Windows 10 and Nvidia published driver updates and I can no longer replicate the crash with the latest updates. I can't really pinpoint what update resolved the issue, sorry for the lack of detail.