realsenseai/librealsense #2650: Only one Infrared stream in Realsense Viewer

stekla79, 2018-11-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435
| Firmware Version                   | 05.09.14.00 
| Operating System & Version |   Win 10  
| Platform                                 | PC  

Hi, 
I'm not able to see the second infrared stream in the Realsense Viewer and the point cloud does not look good. Any ideas to solve this issue? 

[image]

stekla79, 2018-11-02:
I see neither. I'm not able to flip the connector easily since the camera is in a embedded in an enclosure. Any other things to try besides flipping the connector? 

[image]

dorodnic, 2018-11-02:
Hi @stekla79 
You can try uninstalling and reinstalling camera drivers.
I suspect that the lack of `Infrared 2` is a side-effect of the following check:
[win-uvc.cpp:964](https://github.com/IntelRealSense/librealsense/blob/master/src/win/win-uvc.cpp#L964)
Basically, librealsense can't ensure that the camera is connected to USB3, despite the fact that it obvious is. 
This should not limit the list of available resolutions, but can mess-up the defaults and hide some functionality. 
I'd like to understand better what do you mean by "point cloud does not look good"? 

Please navigate to the directory where you have the Viewer, open the command line and enter:
```
set LRS_LOG_LEVEL="DEBUG"
realsense-viewer
```
If everything works well, a new `.log` file will be created in the directory. I'm searching for warnings from [win-helpers.cpp:315-540](https://github.com/IntelRealSense/librealsense/blob/master/src/win/win-helpers.cpp#L315-L540)

RealSense-Customer-Engineering, 2018-11-12:
[Realsense Customer Engineering Team Comment]
Is this issue solved?

closed 2018-11-17 by RealSense-Customer-Engineering

luisfer12, 2021-03-19:
I basically have this error right now. Is there a solution?