realsenseai/librealsense #2656: How to use syncer?

kunyao, 2018-11-01:
### 
I'm trying to use rs2::syncer to group frames from different color sensors from multiple D415. Basically like:
``
sensor0.start(sync)
sensor1.start(sync)
frameset = snyc.wait_for_frames()
``
I found there is only one frame in the frameset returned. Furthermore, when I replace the sensors to color and depth in the same device, it still returns one frame and causes error: null pointer passed for argument "frame_ref" when I want to access the data.

    void testSyncer() {

        rs2::context ctx;
        auto devs = ctx.query_devices();
        rs2::device dev = devs[0];
        auto sensors = dev.query_sensors();                                                                                                                                                                 
        rs2::sensor depth_sensor = sensors[0];
        rs2::sensor color_sensor = sensors[1];
        rs2::stream_profile depth_stream = depth_sensor.get_stream_profiles()[0];
        rs2::stream_profile color_stream = color_sensor.get_stream_profiles()[0];

        depth_sensor.open(depth_stream);
        color_sensor.open(color_stream);

        rs2::syncer sync(2);
        depth_sensor.start(sync);
        color_sensor.start(sync);

        rs2::frameset frames = sync.wait_for_frames();
        std::cout << frames.size() << std::endl;

        rs2::depth_frame depth = frames.first_or_default(RS2_STREAM_DEPTH);
        rs2::video_frame color = frames.first_or_default(RS2_STREAM_COLOR);

        std::cout << depth.get_width() << std::endl;
        std::cout << color.get_height() << std::endl;

Is there anything wrong with the code?

kunyao, 2018-11-02:
Is there anything unclear about the issue? I just copy the code from your document: [https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0](url) but it doesn't work.

kunyao, 2018-11-04:
Finally, figure out how it works after reading the source. Basically, the syncer will release single frame return if it cannot find a matched pair for a certain time(based on timestamp). And the stream[0] for RGB and depth sensor is with different fps.

closed 2018-11-04 by kunyao

dorodnic, 2018-11-07:
Sorry we couldn't provide the information in time. It can take several days for either @RealSense-Customer-Engineering or one of the developers to address incoming issues, especially when we don't have an answer ready like in case of the syncer. You are correct in pointing out this is a gap in SDK documentation. Will try to address soon.

The D400 has somewhat fragile property of `Exposure Priority` control being turned on by default, meaning that in some environments color actual FPS will change. As you already noticed, the syncer will only delay frames that can have potential matches, so in different FPS scenarios it will just let the fast one through. Perhaps we should make this behavior configurable via an option on the syncer object.