realsenseai/librealsense #2660: extract frames from rssdk file

Jilliansea, 2018-11-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello,
I'm new to this project, and I have lots of *.rssdk file, and I want to extract color and depth images frame by frame, what should I do please?

MartyG-RealSense, 2018-11-02:
The 2016 SDK has a tool called Clip Editor that can be used to import recorded SDK files.  you can then analyze, visualize, edit and extract frames.

You can find the tool in the 2016 SDK's Sample Browser application.  there should be a folder the SDK has placed on your desktop called 'Intel RealSense SDK Gold'.  If you open this, there is an executable application inside called 'Intel RealSense SDK Sample Browser'.  

When you run this and the Sample Browser window appears, type 'clip' into the window's search box to quickly find the Clip Editor tool.  Left-click on the 'Run' button to run the program.

[image]

You can access the full documentation for the program by left-clicking on the 'Documentation' button directly below the 'Run' button.

Jilliansea, 2018-11-02:
@MartyG-RealSense  Thank you for your reply very much. Actually I know this GUI, but I have nearly thousands of *.rssdk files, and it's so time-consuming to deal all of them with this GUI one by one. So I want to know whether there are some other ways which can deal a lot of *.rssdk files by a script at once? Or  is there some other convenient means to deal with my problem?

Thanks a lot again.

MartyG-RealSense, 2018-11-02:
I do not know of a pre-made solution for batch-processing .RSSDK files.  I recall that a user once tried to combine RSSDK files together, and was researching the 'struct' format of the files to do so.

https://communities.intel.com/message/522662#522662

philipjames44, 2018-11-02:
@Jilliansea You may be able to create something in visual studio to process files in a folder en masse, but you would have to recreate the code MartyG mentioned to fit your specific use case. I recommend switching to the new SDK, and potentially even switching operating systems to OSX or Ubuntu, and recording new footage. I was in the same position as you with many valuable rssdk files, and updating SDKs not only provided a more flexible way to record and read frames (ROS .bag files), but immensely improved my footage quality. I had so many issues with strange depth flashes that completely vanished once I updated the SDK.

Jilliansea, 2018-11-03:
@MartyG-RealSense Thans a lot. I will search for other ways.

Jilliansea, 2018-11-03:
@philipjames44  Thanks a lot for your advice. My situation is as following, I already have thousands of rssdk files recorded by SR300, but I only have the files and no camera. So my task is to extract color and depth images from the rssdk files and also align them which means I can't record new footage. So I want to know whether there are some ways to extract frames from raw rssdk files without camera? By the way, could you tell me whether the new SDK in Ubuntu is helpful in dealing my problems?
Thanks very much.

MartyG-RealSense, 2018-11-03:
I believe that the .rssdk files are a variant of the video format h.264, also known as MPEG-4 Part 10.  

The link below describes a method for stitching multiple h.264 videos together into one in order of filename, using the program called **ffmeg**.  It says you can use the output of ls/dir to create the sorting list instead of having to index thousands of files manually.
 
https://stackoverflow.com/questions/42686866/merge-all-h264-files-in-directory-in-alphabetical-order
 
Once you have one giant video file, then you could either extract images from the video directly or convert the video into a bag file so it can be used with the current RealSense SDK.  The link below offers a script for doing a video to bag conversion.

https://stackoverflow.com/questions/31432870/ros-convert-video-file-to-bag

Jilliansea, 2018-11-03:
@MartyG-RealSense ok, I will try it. Thanks a lot

zivsha, 2018-11-04:
Note that converting a video to a bag file with the above method will create a bag that cannot be opened using the RealSense Viewer (since it expects a different topic and additional topics other than camera/image_raw.

You could try to create an app that uses both SDKs, the older one to read frames from the file, and the new one to write them to a bag file. You could probably achieve this by creating a synthetic device that reads the file using the older SDK and publish its frames, and then simply use a recording device to wrap the synthetic one. I think that would work

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Hi Jilliansea, How about your test result?

closed 2018-11-14 by RealSense-Customer-Engineering

zhouyongxiu, 2019-02-18:
@Jilliansea I have the some problem, have you solve it?

jinglingzhua, 2019-04-26:
Hello, your problem is actually how to playback rssdk files with code, so checkout these pages, will probably solve your problem:
https://software.intel.com/en-us/blogs/2015/10/15/how-to-record-and-playback-streaming-sequences-in-intel-realsense-sdk
https://gitee.com/Luciferearth/codes/2h6dpv7fwg30qj1ls59ta48
https://stackoverflow.com/questions/10630710/opencv-how-to-get-raw-yuy2-image-from-webcam