realsenseai/librealsense #2661: EntryPointNotFoundException: Entry point not found 'rs2_query_devices_ex' in DLL file 'realsense2'.

eebanos, 2018-11-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Win (10) x64   | 
| Platform                                 | PC |
| SDK Version                            |  2.0 (build 2.16.3)  | 
| Language                               |  C#  |
| .NET Framework                     | 4.6.1 |

### Issue Description
Just updated the Real Sense SDK to the latest release, (uninstalled and perform a clean installation), then on my project reimported the Intel.Realsense.dll for my current release configuration: Release x64.
Imported Intel.Realsense.dll reference from C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64
 Then after compiling VS throws this exception:
EntryPointNotFoundException: Entry point not found 'rs2_query_devices_ex' in DLL file 'realsense2'.

Just for the simple code:
```cs
      public void ExampleTwo()
        {
            FrameQueue q = new FrameQueue();
            using (var ctx = new Context())
            {
                var lst = ctx.QueryDevices();
                foreach (var item in lst)
                {
                    Console.WriteLine(item.Info[CameraInfo.SerialNumber]);
                }
            }
        }
```
How can I tackle this issue??....

The RealSense Viewer works fine and detect both SR300 cameras connected to the PC.

dorodnic, 2018-11-02:
I think there is a version mismatch between `realsense2.dll` and `Intel.Realsense.dll`. I suggest a search for `realsense2.dll` under your .NET solution folder, followed by replacing all occurrences with the one from  `C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64`. Please let me know if it helps, hope it does, this seem like the most likely solution.

eebanos, 2018-11-02:
Solved, just replaced the realsense2.dll on the project folder, not the solution one. Thanks.

closed 2018-11-02 by eebanos