realsenseai/librealsense #2663: Cannot build realsense application with Cmake

ljc19800331, 2018-11-03:
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)   | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { librealsense2 }                          | 
| Language                            |  {C++/pcl/opencv/librealsense }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I tried to build the example application with pcl, opencv and realsense but failed to build the file due to the opengl issue. My CMake file is: 

cmake_minimum_required(VERSION 3.12)
project(Program_1)
set(CMAKE_CXX_STANDARD 14)

find_package(OpenCV REQUIRED)
find_package(PCL 1.3 REQUIRED COMPONENTS common io)

# Find glfw header
find_package(OpenGL)
set(DEPENDENCIES realsense2 ${OPENGL_LIBRARIES})

add_executable(Program_1 main.cpp test_opencv.cpp test_pcl.cpp test_pointer.cpp /home/maguangshen/librealsense/examples/example.hpp)

include_directories(${PCL_INCLUDE_DIRS})
include_directories($(OpenCV_INCLUDE_DIRS))

target_link_libraries(Program_1 ${OpenCV_LIBS})
target_link_libraries(Program_1 ${PCL_COMMON_LIBRARIES} ${PCL_IO_LIBRARIES})
target_link_libraries(Program_1 realsense2)
target_link_libraries(Program_1 ${DEPENDENCIES})

link_directories(${PCL_LIBRARY_DIRS})
add_definitions(${PCL_DEFINITIONS})

And I got the following issue:
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::window(int, int, char const*)::{lambda(GLFWwindow*, int, int, int)#1}::operator()(GLFWwindow*, int, int, int) const':
/home/maguangshen/librealsense/examples/example.hpp:237: undefined reference to `glfwGetWindowUserPointer'
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::window(int, int, char const*)::{lambda(GLFWwindow*, double, double)#2}::operator()(GLFWwindow*, double, double) const':
/home/maguangshen/librealsense/examples/example.hpp:243: undefined reference to `glfwGetWindowUserPointer'
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::window(int, int, char const*)::{lambda(GLFWwindow*, double, double)#3}::operator()(GLFWwindow*, double, double) const':
/home/maguangshen/librealsense/examples/example.hpp:249: undefined reference to `glfwGetWindowUserPointer'
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::window(int, int, char const*)::{lambda(GLFWwindow*, int, int, int, int)#4}::operator()(GLFWwindow*, int, int, int, int) const':
/home/maguangshen/librealsense/examples/example.hpp:255: undefined reference to `glfwGetWindowUserPointer'
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::window(int, int, char const*)':
/home/maguangshen/librealsense/examples/example.hpp:228: undefined reference to `glfwInit'
/home/maguangshen/librealsense/examples/example.hpp:229: undefined reference to `glfwCreateWindow'
/home/maguangshen/librealsense/examples/example.hpp:232: undefined reference to `glfwMakeContextCurrent'
/home/maguangshen/librealsense/examples/example.hpp:234: undefined reference to `glfwSetWindowUserPointer'
/home/maguangshen/librealsense/examples/example.hpp:239: undefined reference to `glfwSetMouseButtonCallback'
/home/maguangshen/librealsense/examples/example.hpp:245: undefined reference to `glfwSetScrollCallback'
/home/maguangshen/librealsense/examples/example.hpp:251: undefined reference to `glfwSetCursorPosCallback'
/home/maguangshen/librealsense/examples/example.hpp:260: undefined reference to `glfwSetKeyCallback'
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::operator bool()':
/home/maguangshen/librealsense/examples/example.hpp:269: undefined reference to `glfwSwapBuffers'
/home/maguangshen/librealsense/examples/example.hpp:271: undefined reference to `glfwWindowShouldClose'
/home/maguangshen/librealsense/examples/example.hpp:273: undefined reference to `glfwPollEvents'
/home/maguangshen/librealsense/examples/example.hpp:274: undefined reference to `glfwGetFramebufferSize'
/home/maguangshen/librealsense/examples/example.hpp:282: undefined reference to `glfwGetWindowSize'
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::~window()':
/home/maguangshen/librealsense/examples/example.hpp:290: undefined reference to `glfwDestroyWindow'
/home/maguangshen/librealsense/examples/example.hpp:291: undefined reference to `glfwTerminate'

I am using Clion. Thanks.

dorodnic, 2018-11-03:
Hi @ljc19800331 
Looking at the Cmake, I don't see any mention of glfw. Only OpenGL, OpenCV, PCL and RealSense. 
Perhaps it is hiding inside `DEPENDENCIES` but I'd double check.
Here is an example of how we pull the dependency in our graphical tools: [tools/CMakeLists.txt:33-64](https://github.com/IntelRealSense/librealsense/blob/master/tools/CMakeLists.txt#L33-L64)
Hope this helps

ljc19800331, 2018-11-03:
@dorodnic Thanks for your reply. I have add these parts in the CMAKE but still get the same errors.

### This is my CMake:
cmake_minimum_required(VERSION 3.12)

# Project name
project(Program_1)

# Set the variables
set(CMAKE_CXX_STANDARD 14)
set(DEPENDENCIES realsense2 ${OPENGL_LIBRARIES})

find_package(OpenGL)

if(WIN32)
    list(APPEND DEPENDENCIES glfw3)
else()
    # Find glfw header
    find_path(GLFW_INCLUDE_DIR NAMES GLFW/glfw3.h
            PATHS /usr/X11R6/include
            /usr/include/X11
            /opt/graphics/OpenGL/include
            /opt/graphics/OpenGL/contrib/libglfw
            /usr/local/include
            /usr/include/GL
            /usr/include
            )
    # Find glfw library
    find_library(GLFW_LIBRARIES NAMES glfw glfw3
            PATHS /usr/lib64
            /usr/lib
            /usr/lib/${CMAKE_LIBRARY_ARCHITECTURE}
            /usr/local/lib64
            /usr/local/lib
            /usr/local/lib/${CMAKE_LIBRARY_ARCHITECTURE}
            /usr/X11R6/lib
            )
    if(APPLE)
        find_library(COCOA_LIBRARY Cocoa)
        find_library(IOKIT_LIBRARY IOKit)
        find_library(COREVIDEO_LIBRARY CoreVideo)
        LIST(APPEND DEPENDENCIES ${COCOA_LIBRARY} ${IOKIT_LIBRARY} ${COREVIDEO_LIBRARY})
    endif()
    list(APPEND DEPENDENCIES m ${GLFW_LIBRARIES} ${LIBUSB1_LIBRARIES})
    include_directories(${GLFW_INCLUDE_DIR})
endif()

# Find the new packages
find_package(OpenCV REQUIRED)
find_package(PCL 1.3 REQUIRED COMPONENTS common io)

# Add executable files
add_executable(Program_1 main.cpp /home/maguangshen/librealsense/examples/example.hpp)

include_directories(${PCL_INCLUDE_DIRS})
include_directories($(OpenCV_INCLUDE_DIRS))

target_link_libraries(Program_1 ${OpenCV_LIBS})
target_link_libraries(Program_1 ${PCL_COMMON_LIBRARIES} ${PCL_IO_LIBRARIES})
target_link_libraries(Program_1 realsense2)
target_link_libraries(Program_1 ${DEPENDENCIES})

link_directories(${PCL_LIBRARY_DIRS})
add_definitions(${PCL_DEFINITIONS})

### These are my errors:
CMakeFiles/Program_1.dir/main.cpp.o: In function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
/home/maguangshen/librealsense/examples/example.hpp:323: undefined reference to `glPopMatrix'
/home/maguangshen/librealsense/examples/example.hpp:324: undefined reference to `glPushAttrib'
/home/maguangshen/librealsense/examples/example.hpp:326: undefined reference to `glClearColor'
[... 32 lines cut ...]
/home/maguangshen/librealsense/examples/example.hpp:112: undefined reference to `glTexImage2D'
/home/maguangshen/librealsense/examples/example.hpp:115: undefined reference to `glTexImage2D'
/home/maguangshen/librealsense/examples/example.hpp:121: undefined reference to `glTexParameteri'
/home/maguangshen/librealsense/examples/example.hpp:122: undefined reference to `glTexParameteri'
/home/maguangshen/librealsense/examples/example.hpp:123: undefined reference to `glTexParameteri'
/home/maguangshen/librealsense/examples/example.hpp:124: undefined reference to `glTexParameteri'
/home/maguangshen/librealsense/examples/example.hpp:125: undefined reference to `glPixelStorei'
/home/maguangshen/librealsense/examples/example.hpp:126: undefined reference to `glBindTexture'
CMakeFiles/Program_1.dir/main.cpp.o: In function `window::operator bool()':
/home/maguangshen/librealsense/examples/example.hpp:268: undefined reference to `glPopMatrix'
/home/maguangshen/librealsense/examples/example.hpp:277: undefined reference to `glClear'
/home/maguangshen/librealsense/examples/example.hpp:278: undefined reference to `glViewport'
/home/maguangshen/librealsense/examples/example.hpp:281: undefined reference to `glPushMatrix'
/home/maguangshen/librealsense/examples/example.hpp:283: undefined reference to `glOrtho'
collect2: error: ld returned 1 exit status
CMakeFiles/Program_1.dir/build.make:252: recipe for target 'Program_1' failed
make[3]: *** [Program_1] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/Program_1.dir/all' failed
make[2]: *** [CMakeFiles/Program_1.dir/all] Error 2
CMakeFiles/Makefile2:84: recipe for target 'CMakeFiles/Program_1.dir/rule' failed
make[1]: *** [CMakeFiles/Program_1.dir/rule] Error 2
Makefile:118: recipe for target 'Program_1' failed
make: *** [Program_1] Error 2

Thanks for your answer.

closed 2018-11-03 by ljc19800331