realsenseai/librealsense #2667: D415 issue while launching

shrijan00, 2018-11-04:
So when i'm launching roslaunch realsense2_camera rs_camera.launch
Even though if i type rqt_image_view, i am able to see rgb and depth images but anybody knows about these error. I guess its either because of kernel issue or sdk.
[image]

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Hi  shrijan00,
Could you provide your running envs(such as  os, platform, camera model, firmware version, kernel version, librealsense SDK version ...)?

Thanks.

shrijan00, 2018-11-13:
Somehow it is working so its ok with me.

closed 2018-11-13 by shrijan00

RealSense-Customer-Engineering, 2018-11-13:
[Realsense Customer Engineering Team Comment]
Thanks your update.