realsenseai/librealsense #2676: Jetson Installation

PioneerAPCS18-19, 2018-11-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15                                         | 
| Platform                                 | Jetson  |
| SDK Version                            |   2.0                           | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description

Install instructions for Linux (which the Jetson installation page says to use) do not work because the Jetson has an ARM processor. Installation instructions from JetsonHacks do not work either. Any chance to have the Jetson officially supported in the future?

MartyG-RealSense, 2018-11-05:
The RealSense 400 Series cameras are compatible with any Intel or ARM processor and have been used successfully with the Jetson boards.  In June 2018 the SDK added CUDA optimizations for Jetson.

https://github.com/IntelRealSense/librealsense/pull/1866

A patch script can be found here:

https://github.com/IntelRealSense/librealsense/issues/1424#issuecomment-375965070

dorodnic, 2018-11-06:
> Any chance to have the Jetson officially supported in the future?

Yes, we are working to improve Jetson support over time. It is not trivial, especially for platforms we have zero control over. Publishing debians for Jetson makes sense. Publishing a DKMS or maintaining the patches for NVidia variant of the kernel, however, is not likely to happen. What we do instead is upstreaming RealSense specific patches to the Linux kernel. With 4.15 you should have almost all hardware features out of the box, except hardware timestamp and frame metadata. Once Jetson distro upgrades to 4.16 you will get hardware timestamp as well.

In addition, as with every other non-Intel non-LTS Linux distro, you have a "backup" backend that does not rely on patching the kernel. To enable it, add `-DFORCE_LIBUVC=true` to CMake (see [Build Configuration](https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration))

We also continue to publish Jetson specific improvements ([#2670](https://github.com/IntelRealSense/librealsense/pull/2670)) based on feedback from the community.

closed 2018-12-14 by RealSense-Customer-Engineering