realsenseai/librealsense #2677: Matlab:Example to "Controlling the Laser" on librealsense2

kklaw85, 2018-11-06:
Hi,
I am currently developing an interface to control D415 on Matlab platform. Just wondering if there is any example to control the laser on librealsense2 with Matlab wrapper? I found an example on C++, but I have difficulty getting it done on Matlab. Here's what I have done so far;

profile = pipe.start();
selected_device=profile.get_device();
depth_sensor=selected_device.first(realsense.sensor());    <- error on this line, Not enough input arguments. I could not figure out what is to be inserted into the realsense.sensor.

This is from the C++ example
rs2::pipeline pipe; 
rs2::pipeline_profile selection = pipe.start();
rs2::device selected_device = selection.get_device();
auto depth_sensor = selected_device.first<rs2::depth_sensor>();

Appreciate it if anyone can guide me along on this. Thank you in advance

lramati, 2018-11-06:
device.first() accepts a string. Try e.g. device.first('sensor') or
device.first('depth_sensor')

kklaw85, 2018-11-06:
Hi Iramati, thank you so much, it works!

closed 2018-11-07 by RealSense-Customer-Engineering