realsenseai/librealsense #2681: Set autoexposure ROI from json or Python

mprat, 2018-11-06:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.3 |
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.14.13                                     | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  2.16.1                     | 
| Language                            |  Python                         | 

### Issue Description

How can I set the autoexposure ROI in the JSON configuration file in advanced mode? I see examples for how to set it in C++ (https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_sensor.hpp#L459) but no documentation for how to change this in the `.json` parameters or from Python. Any help is appreciated!

dorodnic, 2018-11-06:
Hi @mprat 
The ROI is not part of the camera JSON, I do see it is supported in the python wrapper ([python.cpp:686-690](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/python.cpp#L686-L690))
So it should be something like:
```python
p = rs.pipeline()
prof = p.start()
s = prof.get_device().first_roi_sensor()
roi = s.get_region_of_interest()
...
s.set_region_of_interest(roi)
```

dorodnic, 2018-11-29:
Hi @mprat 
Did the code snippet helped?

mprat, 2018-12-04:
Works great, thanks.

closed 2018-12-04 by mprat