realsenseai/librealsense #2683: Pyrealsense2 for exposure adjustment

ljc19800331, 2018-11-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300  } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {Python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to detect the laser spot on the phantom (similar to human skin) but there is a saturation problem for detection (figure 1). I could not detect the centroid of the laser spot based on the RGB image. I changed the exposure from realsense-viewer and the spot is easily detected (figure 2). However, I faced the problem when I used in Python with the code presented in #1011 and #2021 to change the exposure in python. I have  two main questions:
1. Would it possible if I change exposure to a specific value in Pyrealsense2? Is there an easy way to find the corresponding command lines? (I could not easily find in the original code, any example codes would be good).
2. If I want to show both images in the same time, then does it mean I need to change the exposure based on the every time step in order to get the images? Is there any good way to show both images (with different exposure effect) in the same time?
3. Any other suggestion to for laser spot detection is welcome (except for changing the exposure)
[image]
[image]

Thanks.

dorodnic, 2018-11-06:
If I recall correctly, 
```python
import pyrealsense2 as rs
p = rs.pipeline()
prof = p.start()
s = prof.get_device().query_sensors()[1]
s.set_option(rs.option.exposure, new_value)
```
There is no way to make the exposure bounce between two values (at least using librealsense). You need to manually set exposure, wait, get image, set exposure, wait, get another image.

Did you try to enable the Infrared stream? I'm not 100% sure, but there is a chance you can see the dot via the other sensor. And, you can configure its exposure (through `prof.get_device().query_sensors()[0]` and another option, called `motion_range`) I see you are using the SR300, so resolution would not be great, but it is an idea.

dorodnic, 2018-11-27:
@ljc19800331 any luck with your problem?